prusa3d/PrusaSlicer #6724: Slicer did allow yesterday, doesn't allow today.  Same function, same new version.

JasonJelinek, 2021-07-17:
### Version
Version 2.3.1+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
Stock Mini+

### Behavior
Created STL on TinkerCad.  Simple square that fills or almost fills the bed.

Opened STL yesterday.  Found my design was outside of printable area due to skirt.  Unlocked settings and adjusted X,Y,and Z.  It sliced and printed fine last night.  

This morning I opened PSlicer and opened the SAME stl.  Still out of printable area, I unlocked settings and attempted to adjust X,Y, and Z.  Slicer WON'T let me adjust any geometry settings at all, as I had done yesterday on the same STL.

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
[Tremendous Elzing.zip](https://github.com/prusa3d/PrusaSlicer/files/6834885/Tremendous.Elzing.zip)

rtyr, 2021-07-19:
I am not sure I understand. I have no problem after opening your project file and disabling the skirt.

Can you provide some screenshot displaying what exactly do you mean?

rtyr, 2021-09-28:
No response. Closing.

closed 2021-09-28 by rtyr