prusa3d/PrusaSlicer #6726: Start gcode not the real start Gcode

broncosis, 2021-07-19:
### Version

2.3.2

### Operating system type + version
Linux Mint 20

### 3D printer brand / version + firmware version (if known)
custom built Rail core with a duet wifi 

### Behavior
Adding Gcode before start Gcode 

is there a way to turn off the M109 and M104 that prusa adds before the start Gcode 
I use Macros for my start Gcode so that all of my slicers have the same start gcode 
and I only have to change it one place 

It also Causes issues with M80 not being the first and I get heater faults because the printer is not powering on 

this is what Prusa Generates 

; generated by PrusaSlicer 2.3.2+linux-x64 on 2021-07-19 at 04:04:11 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.35mm
; first layer extrusion width = 0.36mm

M107
M190 S70 ; set bed temperature and wait for it to be reached
G10 S210 ; set temperature
;TYPE:Custom
M80
set global.start_bed_temp = 70
set global.start_tool_temp = 210
set global.PA_value = 0.1;
M98 P"0:/macros/Startgcode.g"
G10 S210 ; set temperature
M116 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
; Filament gcode
set global.PA_value = 0.15
;LAYER_CHANGE
;Z:0.3
;HEIGHT:0.3
G1 Z0.300 F24000.000

I assume the slicer is looking for the M109 and M104 commands in the start Gcode and if not found it adds 
its own 

this is my start G code
M80;
set global.start_bed_temp = [first_layer_bed_temperature];
set global.start_tool_temp = [first_layer_temperature];
set global.PA_value = 0.1;
M98 P"0:/macros/Startgcode.g";

as you can see it keeps my gcode very simple and portable between my few machines
for the most part 
I know I am probably asking something that is not really required for a Prusa printer 
but just a check box to stop what ever assumptions its making about the start Gcode 

#### Project File (.3MF) where problem occurs   
renamed to .txt because github does not support 3mf
[vaccumstrng90pipe.txt](https://github.com/prusa3d/PrusaSlicer/files/6838157/vaccumstrng90pipe.txt)

broncosis, 2021-07-19:
I made a work around by adding a blank M190  and M104 to the start gcode 
would still be nice

rtyr, 2021-07-19:
Already requested in https://github.com/prusa3d/PrusaSlicer/issues/2420. I am closing this as a duplicate.

closed 2021-07-19 by rtyr