prusa3d/PrusaSlicer #6732: Underextrusion following an M600 colour change

MattBlissett, 2021-07-21:
### Version

PrusaSlicer-2.3.2+linux-x64-202107080710.AppImage

### 3D printer brand / version + firmware version (if known)

Prusa i3 Mk3S, firmware version 3.10.0.  No MMU.

### Behavior

After an M600 filament change, prior to the fix in #6362 there was a blob.  After that fix, we seem to have underextrusion for about 3-5mm – see the image, showing some white layers, then a colour change and a single black layer.

Where the colour is only a single layer thick (sign, board game etc) this leaves a visible flaw in the print.

#### Project File (.3MF) where problem occurs

[M600-Test.zip](https://github.com/prusa3d/PrusaSlicer/files/6854891/M600-Test.zip) (a cylinder with no infill or top/bottom layers, and a colour change for the top layer).

[image]

[image]

ssill2, 2021-07-21:
I've seen something similar even with master builds recently on the first layer of the wipe tower with mmu print.  After it infills with the first color and then does a normal mmu color change.   Where it would normally draw the perimeter and then the brim of the wipe tower in the second filament, 1/2 to 3/4 of the perimeter is either not extruded at all or looks like the picture above.  By the time it comes around to start on the wipe tower's brim the extrusion is back to normal.

btw, in the pic above your zOffset seems too high.  you can see the gap between the extrusions.

rtyr, 2021-07-21:
@MattBlissett 

I will test it, but the g-code looks OK. Retract -> M600 (nozzle is primed after filament change) -> Continue printing. M600 behavior itself is handled completely by the firmware.

[image]

@ssill2 

> I've seen something similar even with master builds recently on the first layer of the wipe tower with mmu print. After it infills with the first color and then does a normal mmu color change. Where [...]

Not related to this issue.

rtyr, 2021-08-02:
The g-code is OK, I was able to reproduce the issue though. For now, I will likely add partial "deretraction" after M600 (for our printers), it can be added in printer profile (custom g-code section). We may somehow improve the M600 behavior on firmware side in the future.

rtyr, 2021-09-13:
Nozzle priming after M600 was added in recent configuration update 1.3.2. It should at least partially help with this issue.

closed 2021-09-13 by rtyr