prusa3d/PrusaSlicer #6736: "solid_infill_every_layers = 1" does not generate valid infill for some geometries

el-bart, 2021-07-22:
### Version
PrusaSlicer-2.3.3+linux-x64-202107161032 based on Slic3r (with GUI support)

### Operating system type + version
Linux Debian 10.9

### 3D printer brand / version + firmware version (if known)
N/A (problem visible on a gcode preview)

### Behavior
* _Describe the problem_
  * 100% infill is missing on some layers, when `solid_infill_every_layers` is set to sth else than `0`.
  * happens on certain models only.
  * small changes in the model structure can make the problem appear or go away.
* _Steps needed to reproduce the problem_
  * ./AppRun --load   config.ini --output model.gcode --export-gcode model.stl
  * both `config.ini` and `model.stl` are in `tar.gz`, linked below.
* _Expected Results_
  * infill is 100% on all layers when `solid_infill_every_layers = 1`
* _Actual Results_
  * no infill is present on all layers, except of bottom and top ones.

[image]

note that problem goes away, when `solid_infill_every_layers = 0` - then proper infill pattern is created:

[image]

this is a problem inherited from Slic3r. reported here:
https://github.com/slic3r/Slic3r/issues/5091

#### Project File (.3MF) where problem occurs

problem can be reproduced from command line, with a following `config.ini`, `model.stl` and `./generate` script:
[solid_infill_issue.tar.gz](https://github.com/prusa3d/PrusaSlicer/files/6860701/solid_infill_issue.tar.gz)
i've also included `model.scad` from where `model.stl` is generated. this problem looks like a corner case, as small changes in the model (eg. holes layout/density) make it go away.

bubnikv, 2021-12-10:
fixed with

bubnikv, 2021-12-10:
fixed with a132dddb8b3320464a4701fd712a8000abd2a314

    With solid_infill_every_layers = 1 the new solid layer should be solid
    infill not solid bridging infill. Once the layer is correctly classified
    as solid with solid_infill_every_layers = 1, everything works as expected.

closed 2021-12-10 by bubnikv