prusa3d/PrusaSlicer #6737: PrusaSlicer does not render edges

jkuusama, 2021-07-22:
### Version
Version 2.3.3+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3S+, latest firmware

### Behavior
Part appearance in Prusaslicer:
[image]

Same part in Fusion 360:
[image]

murk-sy, 2021-07-22:
It looks simple yes, but Prusaslicer does not actually have the same model that you're looking at in Fusion.

What you see in Fusion (though examples are from Sketchup):
[image]

What Prusaslicer gets:
[image]

Prusaslicer does not by itself have a concept of what is a single surface and what isn't.

However, I agree that improving rendering would be nice. Something like Sketchup's "Profiles" style might look good while being relatively easy to implement if they decide not to actually show meshes themselves. It's basically kinda like cell shading. and it allows you to see gaps in models.
[image]

jkuusama, 2021-07-22:
It might be as simple as calculating the angle of between the triangles on each side of the dotted line and if bigger than a threshold, it is an edge, so draw a line.

kkozlik, 2026-09-25:
I would say this is much improved in the 3.0 version. But still not ideal:

[image]