prusa3d/PrusaSlicer #6738: feature req - add cone shape (next to box cylinder and sphere)

pixsrv, 2021-07-22:
new feature request
please add cone shape next to existing shapes

properly scales and positiones cones could be used as support shapes
similar to meshmixer support trees

in the simplest form user could just add a cone on the bed
and scale it the way it will reach and support overhangs

or
coupled with cylinders the user could form regular support tree

or
there could be kinda special object like cylinder-with-cone-on-top
to mimic support trees

but the cone could just do the job

as for now it's possible to mimic such support this way
[image]
but cones coud do it better

rtyr, 2021-07-22:
We added new "shape gallery", allowing you to add your own shapes. It will be included in upcoming 2.4.0-alpha1. 

Similar to https://github.com/prusa3d/PrusaSlicer/issues/5275. Closing.

closed 2021-07-22 by rtyr

bubnikv, 2021-07-26:
Modifier gallery implemented in the upcoming PrusaSlicer 2.4.0-alpha1.