prusa3d/PrusaSlicer #6739: Some FDM supports fail to generate for small polygons.

n8bot, 2021-07-22:
### Version
2.4.0 Alpha compiled from latest master branch.

### Operating system type + version
Win 10

### Behavior
Painted- on supports, as well as auto-generated supports fail to generate on some models/in some areas.

[image]

[image]

It seems related to the very high mesh resolution of the STL. The resulting polygons are tiny, and I think that is preventing the supports from generating.

In 2.3.0 alpha I encountered a similar problem, which was solved by this commit: 84595ba22bc86084f3cabe8169b50e7642463b8a

Unfortunately that no longer solves the issue. The whole thing has been refactored, as you know, and I now know less about how it works than I did before... and I knew nothing before.

#### Project File (.3MF) where problem occurs
[Troubleshooting2.zip](https://github.com/prusa3d/PrusaSlicer/files/6863636/Troubleshooting2.zip)

The 3mf is currently set for snug supports, but that was just for trial. With the normal supports it doesn't work either. In 2.3.0 this exact support painting generates perfectly all over with the commit above.

Some of the other settings might not make sense. This is just for illustration purposes. E.g., ignore the weird height skirt.

n8bot, 2021-07-30:
I mentioned that previously a commit 84595ba22bc86084f3cabe8169b50e7642463b8a fixed this, but no longer had any effect.

I've discovered that this is due to the conditional on this line:

https://github.com/prusa3d/PrusaSlicer/blob/2f6f1f0e5555e8c41daef9c4f4a0fe033282c9c2/src/libslic3r/SupportMaterial.cpp#L1446

I presume that the support_auto flag is no longer being set like it was before. Anyway... I'd love to see or hear about any of the ideas the prusa team has for improving the supports. They are... complicated to say the least.

n8bot, 2021-08-03:
Ok, I actually figured something out here.

I don't know why I was having issues with auto supports generating, those seem to work. Snug have a few missing layers and such, but grid works ok.

Enforcers don't work. I changed a line of code, and finally the enforcers generated supports!

```cpp
            enforcer_polygons = annotations.enforcers_layers[layer_id];//diff(intersection(layer.lslices, annotations.enforcers_layers[layer_id]),
                // Inflate just a tiny bit to avoid intersection of the overhang areas with the object.
                //offset(lower_layer_polygons, 0.05f * fw, SUPPORT_SURFACES_OFFSET_PARAMETERS));
```

I commented out some things, and made it so `enforcer_polygons` is only `annotations.enforcers_layers[layer_id]` 

Result: 
[image]

I think the triangle selector is not keeping the translation of the model correct, or something. That's why the intersect() doesn't work. It can't intersect!

n8bot, 2021-08-03:
Okay, so I discovered a way to resolve this issue. It's indeed a problem with the translation, but in PrintObject.cpp project_and_append_custom_facets():

https://github.com/prusa3d/PrusaSlicer/blob/9b1906033441dab648fc3236131faf1cedcaffff/src/libslic3r/PrintObject.cpp#L2301

Changing that line as such:

```diff
void PrintObject::project_and_append_custom_facets(
        bool seam, EnforcerBlockerType type, std::vector<Polygons>& out) const
{
    for (const ModelVolume* mv : this->model_object()->volumes)
        if (mv->is_model_part()) {
            const indexed_triangle_set custom_facets = seam
                    ? mv->seam_facets.get_facets_strict(*mv, type)
                    : mv->supported_facets.get_facets_strict(*mv, type);
            if (! custom_facets.indices.empty())
                project_triangles_to_slabs(this->layers(), custom_facets, 
-                    (Eigen::Translation3d(to_3d(unscaled<double>(this->center_offset()), 0.)) * this->trafo() * mv->get_matrix()).cast<float>(), 
+                    (Eigen::Translation3d(Vec3d(-unscale<float>(m_center_offset.x()), -unscale<float>(m_center_offset.y()), 0)) * this->trafo() * mv->get_matrix()).cast<float>(),
                    seam, out);
        }
}
```

Now the enforcer supports work perfectly!

n8bot, 2021-08-09:
I notice that the reason my changed code works, is because of the minus sign before the function calls. The translation of X and Y needs to be negated to place them back from wherever they've been translated to or something.

bubnikv, 2021-08-10:
Thanks, there was certainly a bug in application of the centering offset in PrintObject::project_and_append_custom_facets() after support generator refactoring. 

[image]

Fixed with 04d6b17c572bf732292b84bca875cf2ccb406efb

closed 2021-08-10 by bubnikv