prusa3d/PrusaSlicer #6741: Outer perimeter doesn't always follow on from inner perimeter

iso250, 2021-07-23:
### Version
2.3.3

### Operating system type + version
OSX 11.4

### 3D printer brand / version + firmware version (if known)

Mk3s

### Behavior

I print a lot of 2 perimeter thin walled objects using large nozzles.  When the outer perimeter follows on from the inner perimeter the outer seam looks nice and neat.  I'm not sure why but on some objects the slicer will plot the course of the inner perimeter, retract, then move the extruder to another part of the print and then start the outer perimeter.  

Not only does this cause neatness issues for larger nozzle prints but it's also extremely inefficient.

This issue happens when you use aligned or nearest seam positioning.

n8bot, 2021-07-23:
What you are describing sounds like the opposite of the intended behaviour.

The slicer should plan the extrusion as grouped into islands. So, each little island will have all parts of it printed before moving onto the next island.

What you are describing sounds like the opposite of that: printing the internal perimeters of one island, and before completing all the extrusion on that island, moving to another one. Is this correct?

iso250, 2021-07-23:
It is completing islands one at a time but in a disjointed manner. 

Imagine its printing a cube with 2 perimeters. It starts printing the inner perimeter at the front of the cube to the left and prints counter-clockwise. As soon as the extruder reaches the start point of that perimeter (the front left) instead of continuing on to the outside perimeter it is retracting, moving to the rear right and then starting the outside perimeter. 

It’s inefficient and can cause seam issues.

n8bot, 2021-07-23:
I understand now what you mean! I'm unaffiliated with prusaslicer/prusa team, so you'll have to wait a bit more for someone with expert knowledge. I was just trying to get further clarification. Thank you!

I can see how that could be unwanted.

iso250, 2021-08-12:
Anyone from Prusa had a chance to look at this yet?

bubnikv, 2021-08-12:
@iso250 could you please post pictures here? It will saves us a lot of time.

iso250, 2021-08-12:
So let's use this as an example.  This is seam set to aligned.

[image]

When printing the inner walls you can see where the line begins.

[image]

When it completes the inner perimeter it continues on to do the outer perimeter.  That's how it should work.

[image]

When you print multiples of the same object then issues appear.  The first object prints in the manner I outlined above.  The second object has errors in the path.  Here, the first object has printed as it should.  Now with teh second object it has started the inner perimeter...

[image]

It completes the inner perimeter...

[image]

But then it retracts and moves to another point...

[image]

And then unretracts and starts the outer perimeter.

[image]

iso250, 2021-09-22:
I logged this issue before the launch of the previous incarnation of PrusaSlicer.  The issue wasn't fixed.  Now it seems that this version of PrusaSlicer won't be addressing this.

99.9% of all prints have perimeters.  Perimeters are one of the most important paths that a slicer needs to consider and this issue shows that there is a problem with how the perimeter path is calculated.

Instead of this even being looked at they add a "feature" whereby the slicer recognises when a print may be a sign.  Seriously, 55 of people may use that once or twice in their printing life, yet practically every print of theirs will have perimeters.  Absolutely insane.

dartrax, 2021-09-23:
> When that didn’t fix it I have to use superslicer as the problem didn’t exist there. The thing is I HATE super slicer. It doesn’t produce as nice a finish as PrusaSlicer.

I think @supermerill would be interested in why superslicer doesn‘t produce the same nice finish for you. Improving that could be another possible solution for you. As you may know, Superslicer has a feature that connects the perimeters into one continuous loop, perfect for leaky nozzles!

iso250, 2021-09-23:
> I think @supermerill would be interested in why superslicer doesn‘t produce the same nice finish for you. Improving that could be another possible solution for you. As you may know, Superslicer has [...]

I don't know why but I've found that top layers on PrusaSlicer seem smoother.  Yes, you can always iron but for what I print ironing would make the prints take a lot longer than I want.

Interestingly enough, the one loop perimeter isn't needed for what I print as SuperSlicer doesn't have the issue I' made this ticket about.