prusa3d/PrusaSlicer #6744: Manually edited SLA supports are shifted below the object

davidkerny, 2021-07-24:
2.3.3 Windows 10, SL1

Automatically generated supports are attached to the object but manually added supports are shifted below the object, some even are below the platform.

[image]
[supports below platform.zip](https://github.com/prusa3d/PrusaSlicer/files/6872540/supports.below.platform.zip)

tamasmeszaros, 2021-08-19:
Hi @davidkerny! It seems like the 3mf project file is somehow corrupted. If I save the object itself as stl, then reload it in a clean project, generate support points then add some manual support points to it, the resulting 3mf project file after save is completely usable. I can't really make the saved project corrupt in a similar way to your example. Do you remember anything you did before saving the project? There seem to be no other config change or transformation to the model. Did you save the project in 2.3.3 or in an older release?

lukasmatena, 2021-08-24:
Possibly related to #6100

davidkerny, 2021-08-24:
Yes, related to #6100. This is just how meshes exported from Fusion 360 behave in general. Saving the file as STL and importing it again works.

lukasmatena, 2021-09-24:
Should be fixed by 6b25a9c for PrusaSlicer 2.4.0-alpha2. Please, retest after it is released. Thanks.
Closing.

lukasmatena, 2021-09-24:
Sorry, correction. The fix is buggy, it will not be part of 2.4.0-alpha2.

stepikovo, 2022-03-14:
I'm closing it as a duplicate of #6100
This issue should be fixed in 2.5.0

closed 2022-03-14 by stepikovo