prusa3d/PrusaSlicer #6746: Support generated inside nested object

jmd73, 2021-07-25:
### Version
version_2.4.0-alpha0-819-g9154ebb16
9154ebb165670fc86da979a88afc56e4d24f1976
(2.3.0 release version works better. but is not perfect, see image 2 below.)

### Operating system type + version
Windows 10 Pro 64bit

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S and MK2.5S

### Behavior
* _Describe the problem_
Support from build plate only active or everywhere. When having a nested object, the supports are generated inside/over the nested object.
(See image 1 below)

* _Steps needed to reproduce the problem_
Put one object inside another and activate supports.

* _Expected Results_
The supports should not be generated inside the nested object. (Over the object should probably be acceptable.)
* _Actual Results_
Supports go through the nested object.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Image 1 (2.4.0-alpha 0 built from source):
[image]

Image 2 (2.3.0 release version, 64bit):
[image]

#### Project File (.3MF) where problem occurs
Project files from both 2.3.0 and 2.4.0-alpha0 provided below.
[nesting_with_supports_PrusaSlicer_2_3_0.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6873594/nesting_with_supports_PrusaSlicer_2_3_0.3mf.zip)
[nesting_with_supports_PrusaSlicer_2_4_0-Alpha0.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6873595/nesting_with_supports_PrusaSlicer_2_4_0-Alpha0.3mf.zip)

neophyl, 2021-07-25:
This is down to the fact that PS slices all objects individually.  If you don't want it to do that then you must merge your objects into one.  This can be done by selecting multiple objects and right clicking and using the merge option.
By slicing individually it allows completely independent layer heights etc for each object.

jmd73, 2021-07-30:
@neophyl Thanks, I didn't think about that merging.
But if I merge the objects, there will not be any support material generated above the inner object. So the dome will not have any support material in the center. And I guess the problem will be greater if the nested object is bigger.

I wrote the issue since there have been talk a long time ago, that the auto arrange function should be able to nest objects in the future, and this can be a problem for new users.

And also, the 2.3 version handles the nesting better (but not perfectly).

@bubnikv What do you think? Should I close this issue with the "merging"-solution is the way to solve this, or should the slicer handle this case differently?

It really isn't a problem for me now, but I thought it could be a problem for others.

bubnikv, 2021-07-30:
I agree with @neophyl, you should merge the objects into one if they overlap in projection to the XY plane.

> But if I merge the objects, there will not be any support material generated above the inner object. So the dome will not have any support material in the center.

That should not be the case. Do you possibly have the "support on print bed only " option enabled?

jmd73, 2021-07-30:
@bubnikv I don't remember if I tested with "supports everywhere" orbuild plate only when I tested after the merge. I will test it when I get home from work and post the result.

jmd73, 2021-07-30:
@bubnikv You were right. I forgot to test it with supports everywhere. When supports everywhere are active and the  objects are merged, it slices as expected.

I am closing this issue.

closed 2021-07-30 by jmd73