prusa3d/PrusaSlicer #6747: Prusa Slicer crashes on open on Ubuntu 21.04

nhendin, 2021-07-26:
### Version
PrusaSlicer-2.3.3+linus-x64-202107161032.AppImage

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Ubuntu 21.04

### 3D printer brand / version + firmware version (if known)
MK3S (not applicable here_

Upon opening the app image, I get a splash screen and the following error message in the console, which does not leave me much to go on.
neil@orac:bin$ ./PrusaSlicer-2.3.3+linux-x64-202107161032.AppImage 
Gtk-Message: 17:21:12.020: Failed to load module "canberra-gtk-module"

(prusa-slicer:658929): Gdk-WARNING **: 17:21:12.048: shmget failed: error 28 (No space left on device)
The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 598 error_code 2 request_code 151 minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

nhendin, 2021-07-26:
If I try the debian package that now is available I get a similar error:
neil@orac:bin$ whereis prusa-slicer
prusa-slicer: /usr/bin/prusa-slicer /home/neil/bin/prusa-slicer /usr/share/man/man1/prusa-slicer.1.gz
neil@orac:bin$ /usr/bin/prusa-slicer

(prusa-slicer:661159): Gdk-ERROR **: 17:41:55.580: The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 343 error_code 2 request_code 151 (GLX) minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
Trace/breakpoint trap (core dumped)

trevjonez, 2021-09-13:
getting similar.
```bash
➜  Downloads ./PrusaSlicer-2.4.0-alpha1+linux-x64-202109011425.AppImage 
[2021-09-13 13:51:18.164794] [0x00007f2aeae358c0] [trace]   Initializing StaticPrintConfigs
Gtk-Message: 13:51:18.235: Failed to load module "canberra-gtk-module"
The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 624 error_code 2 request_code 151 minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
➜  Downloads ./PrusaSlicer-2.4.0-alpha1+linux-x64-GTK3-202109011434.AppImage 
[2021-09-13 13:51:38.593834] [0x00007f04179fd8c0] [trace]   Initializing StaticPrintConfigs

(prusa-slicer:66075): Gdk-ERROR **: 13:51:39.047: The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 326 error_code 2 request_code 151 (GLX) minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
[1]    66075 trace trap (core dumped)  ./PrusaSlicer-2.4.0-alpha1+linux-x64-GTK3-202109011434.AppImage
```

trevjonez, 2021-09-13:
possibly more helpful error message on 2.3.3 builds.
```bash
➜  Downloads ./PrusaSlicer-2.3.3+linux-x64-202107161032.AppImage            
Gtk-Message: 13:55:23.409: Failed to load module "canberra-gtk-module"

(prusa-slicer:66743): Gtk-CRITICAL **: 13:55:37.718: IA__gtk_window_resize: assertion 'height > 0' failed
The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 2934 error_code 2 request_code 151 minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
➜  Downloads ./PrusaSlicer-2.3.3+linux-x64-GTK3-202107161044.AppImage 

(prusa-slicer:66979): Gtk-CRITICAL **: 13:56:25.614: gtk_window_resize: assertion 'height > 0' failed

(prusa-slicer:66979): Gdk-ERROR **: 13:56:25.724: The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 24632 error_code 2 request_code 151 (GLX) minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
[1]    66979 trace trap (core dumped)  ./PrusaSlicer-2.3.3+linux-x64-GTK3-202107161044.AppImage
```

lukasmatena, 2021-09-14:
Does the GTK3 AppImage work?

trevjonez, 2021-09-14:
not for me no. each of those posts is me trying both versions of the app image. is there anything I can do to produce more helpful output when launching from the CLI?

lukasmatena, 2021-09-15:
@trevjonez This exact issue was already reported several times and people fixed it by rebooting their system (apparently it broke after drivers were updated). That might be worth a try. If it does not help, it will probably not be exactly simple to debug.

trevjonez, 2021-09-25:
`PrusaSlicer-2.4.0-alpha2+linux-x64-GTK3-202109241338` worked fine for me today.  not sure if my system got updates or if yall fixed it here.

lukasmatena, 2021-09-26:
@trevjonez Thanks for the update.

@nhendin Does the 2.4.0-alpha2 work for you as well?