prusa3d/PrusaSlicer #6750: "Avoid crossing perimeters" is not working

Kaelum, 2021-07-26:
### Version
2.3.3

### Operating system type + version
Windows 10 21H1

### 3D printer brand / version + firmware version (if known)
Snapmaker 2.0 A350 v1.13.1 (custom settings, nothing inherited)

### Behavior
Travel moves are ignoring perimeters and gouging the tops of prints.  I first noticed this in version 2.3.1.

rtyr, 2021-07-26:
Please provide your 3mf project file and screenshots where do you think it is not working.

Kaelum, 2021-07-26:
Here's the .3mf zipped:
[Tippi Tree Leaf Kit (12).zip](https://github.com/prusa3d/PrusaSlicer/files/6881342/Tippi.Tree.Leaf.Kit.12.zip)

And here is a screen shot of the travel paths:
[image]

rtyr, 2021-08-10:
Intended behavior. The new algorithm tries to route a travel path inside object islands with a shortest path while avoiding holes and concavities, while it **takes the shortest path when traveling between islands**.

See https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-beta3.

Closing.

closed 2021-08-10 by rtyr

Kaelum, 2021-08-10:
@rtyr so, you're basically telling us to use Cura, and to stop using PrusaSlicer.  "Avoid crossing perimeters" is supposed to take longer travel paths.  That is it's SOLE purpose. to ensure that the nozzle does not move over any models while traveling.

It was nice while it lasted, but back to Cura.