prusa3d/PrusaSlicer #6756: Cannot Color Change (M600) at height when MMU Painting

chiefspecialk, 2021-07-28:
### Version
2.4.0-Alpha0 compile from commit a54f5fb41a5d9c4301d217490c086ca6ae208246

### Operating system type + version
Windows 10 x64 build 19043.1110

### Behavior

When using MMU painting unable to add a color change at height to any extruder except the primary/default as it does not see the other extruders as used.

[image]

Trying to add color change at height to extruder 2 and re-slicing with color change gives the following error.

[image]

This feature works if the model is separated and just extruder/colors are assigned to parts as then the extruders will show (used) in the color change selection.  Seems to just be an issue when using the MMU painting feature was it doesn't mark the extruders as used.

bubnikv, 2021-08-06:
Thanks, we are able to reproduce it. @YuSanka is looking into it.

bubnikv, 2021-08-11:
Fixed in current master.

closed 2021-08-11 by bubnikv

kylekyle, 2022-02-02:
I did some MMU painting near the bottom of my print and then tried to do a color change for the last few layers. A color palette pops up and then I get the error pictured above. If I use the paint tool to paint everything instead of a color change at height, it works great. The result is identical, it's just a lot more work to paint every facet. 

I'm running 2.4.1-beta1+x64 on MacOS.

FidelCapo, 2022-02-03:
@kylekyle Can you please attach your project file as zipped 3mf. Thank you.