prusa3d/PrusaSlicer #6758: Not able to export directly to mounted devices

Masterv98, 2021-07-29:
### Version 2.3.3

### OS: Fedora 34 Workstation (literally out of the box) (Linux)

### Biqu B1 modded (og E3D V6) with modded marlin 2.0.9.1

### Behavior
After slicing, when you go to export the Gcode, it opens a new window with a file explorer so you can export. Although it only shows your home directory, not any thumbstick or anything. If you select "other locations", the file explorer will crash. Even though I can access everything directly with the file explorer.
I've been exporting to a random folder to then move to my thumb drive. It's not a major bug, but it's annoying. I tried on the windows version (even though it takes longer to slice there because, windows....) the export directly to an external drive is much easier.

#### 
[image]

neophyl, 2021-07-29:
Related to https://github.com/prusa3d/PrusaSlicer/issues/4772 ?

combolek, 2021-08-02:
I am on Fedora 34 so I tried it with the 2.3.3 AppImage and could not reproduce it. It works either by clicking on "Export G-code" and browsing to the USB mount directory or by clicking on the SD/USB icon to the right of it which takes me directly to the mount directory. Either way I can save the g-code file and there is no crash.

Masterv98, 2021-08-03:
Great, gonna try the appimage (even thought I prefer the installed version through rpm or flathub).

xarbit, 2021-09-02:
there is a update building for PrusaSlicer 2.3.3 with more permissions. I hope this fixes the issues. 

https://flathub.org/builds/#/builders/32/builds/58287

xarbit, 2021-09-02:
@Masterv98 Please report back when you where able to update so this ticket can be closed.

Masterv98, 2021-09-04:
> @Masterv98 Please report back when you where able to update so this ticket can be closed.

YES, IT WORKS NOW!!!! THANK YOU GUYS SO MUCH!!

xarbit, 2021-09-04:
@bubnikv this is solved

bubnikv, 2021-09-06:
Great, thanks!

closed 2021-09-06 by bubnikv