prusa3d/PrusaSlicer #6760: Placeholder/Variable evaluation issue

TimW55, 2021-07-31:
### Version
Version:   2.3.3+win64
Build:     PrusaSlicer-2.3.3+win64-202107161027

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Printer: Prusa MK3S stock model.
Firmware: 3.10.0-4481

### Behavior
Placeholder/Variable evaluation not working in all situations.
I have created a very simple model and modified the default Printer Settings/Custom G-codes.  It seems that evaluation of Placeholders occurs correctly for:

Start G-code
End G-code
Before layer change G-code
After layer change G-code

But there is no evaluation of Placeholders for:

Color Change G-code
Pause Print G-code
Template Custom G-code

When not evaluated the exported G-code contains the Placeholder.

#### Project File (.3MF) where problem occurs
[Placeholder_Eval.zip](https://github.com/prusa3d/PrusaSlicer/files/6911106/Placeholder_Eval.zip)
Please note that this model has been created to illustrate the problem.  The custom g-code is not intended to be used for printing a real model.

combolek, 2021-08-02:
I think this a duplicate of #5782 which was fixed on master in 224f3e506c9db3d2de9e2e441ee4443de2bab964. But there have been several duplicate reports already so perhaps the fix could be considered for cherry-picking into stable.

TimW55, 2021-08-02:
This does indeed seem to be a duplicate of #5782.  I hope that a fix can be made in the next stable release.
Thanks for your help.

rtyr, 2021-08-06:
Fixed on master. It will be included in the upcoming PrusaSlicer 2.4.0-alpha1. Closing.

closed 2021-08-06 by rtyr

TimW55, 2024-03-26:
This issue has reappeared in PrusaSlicer version 2.7.2.  I have recently upgraded my printer from MK3S+ to MK3.5 but I don't think that this issue is sensitive to the printer model.