prusa3d/PrusaSlicer #6761: M900 is sent twice in a row before print starts both LA 1.5 and LA 1.0 K values

gudnimg, 2021-07-31:
### Version
2.3.3

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
* Printer: MK3S+
* Firmware: 3.10.1-RC1

### Behavior
* M900 is sent twice when print starts.
Gcode:

```gcode
M900 K0.05 ; Filament gcode LA 1.5
M900 K30 ; Filament gcode LA 1.0
```

This seems error-prone to me. The slicer should know which printer only supports LA 1.0 and send a value of 30. Then send a 0.05 for those that support LA1.5.

In my opinion the gcode should only be sending:

```gcode
M900 K0.05 ; Filament gcode LA 1.5
```

If the second value is <= 10 then it will overwrite the LA 1.5 value which is a big concern.

The firmware only knows that when the K value is > 10, then it will activate LA 1.0, if it is < 10 it is LA 1.5

* _Steps needed to reproduce the problem_
Just slice any 3D model and view the gcode
* _Expected Results_
* Only one M900 command to supply one K value
* _Actual Results_
Two M900 commands are sent in a row with different K values.
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Here is a shot of my Serial Stream. Note that I have added messages to make more clear what is happening in the code

[image]

#### Project File (.3MF) where problem occurs
* Project file: [crossing.zip](https://github.com/prusa3d/PrusaSlicer/files/6911214/crossing.zip)

gudnimg, 2021-07-31:
I suspect the issue is in this custom G-code
[image]

rtyr, 2021-07-31:
That is intended, this way it's compatible with printers running old firmware (pre 3.9.0) and also MK2S. Current MK2.5/MK3 firmware will automatically pick correct value (it will ignore second LA1.0 value). 

Closing.

closed 2021-07-31 by rtyr

rtyr, 2021-07-31:
See 3.9.0 release notes for more information. https://github.com/prusa3d/Prusa-Firmware/releases