prusa3d/PrusaSlicer #6763: Broken Modifier "Layers and Perimeters" Settings

EventHorizonResearch, 2021-08-01:
PrusaSlicer 2.3.3

Windows 10

### Behavior
* Modifier Bottom Layers is broken.  In order to set an amount of bottom layers, you need to specify the layer number of where you want the bottom layers to end instead of how many you want.  Example:  I want 4 bottom layers for a modifier that begins at layer 26.  I would need to set bottom layers to 29 in order to get the right amount of bottom layers instead of just putting 4 in the "Bottom Layers" input box.  This is unique to bottom layers of modifiers and perimeters and top layers are not also affected by this problem.

[image]
[image]

rtyr, 2021-08-01:
Please provide your 3mf project file.

EventHorizonResearch, 2021-08-01:
[OutletCoverStandard.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6913292/OutletCoverStandard.3mf.zip)

rtyr, 2021-08-02:
I don´t see any issue. There is no bottom surface at layer 26, so it doesn´t make much sense to add modifier for it. If you want solid surface in that area, just use height range modifier with 100% infill.

[image]

EventHorizonResearch, 2021-08-02:
So what you're saying is that modifiers don't create layers at they're boundaries?

neophyl, 2021-08-02:
Nope they don’t. They can under certain conditions create perimeters but they don’t create top or bottom layers. Both Bottom surfaces and Top surfaces/ layers have rather specific criteria from what I can discern and when they are classed in that way then all the other parameters that apply to that type are also applied like ironing if enabled for example.  It’s why the top fill infill pattern is only applied to the very topmost layer rather than say all of the top layers. Anything inside a model is just infill and gets treated differently.

closed 2021-08-02 by EventHorizonResearch