prusa3d/PrusaSlicer #6764: Software crashes when slicing

Caraffa-git, 2021-08-01:
### Version
2.3.1

### Operating system type + version
Manjaro Linux KDE Plasma (5.22.4)
Kernel 5.11.4-1-rt11 (64-bit)

### Behavior
Every time when i click "Slice now" slicer crashes. Here is terminal output
`$ prusa-slicer
WARNING: cgroup v2 is not fully supported yet, proceeding with partial confinement
Gtk-Message: 17:04:56.070: Failed to load module "xapp-gtk3-module"
Gtk-Message: 17:04:56.125: Failed to load module "colorreload-gtk-module"
Gtk-Message: 17:04:56.125: Failed to load module "appmenu-gtk-module"

(prusa-slicer:17966): dbind-WARNING **: 17:04:56.130: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-5tqHjV8gDS: No such file or directory

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.508: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.509: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.509: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.509: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.509: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.509: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.509: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.509: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.757: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.757: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.757: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.757: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.757: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.757: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.757: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.758: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.758: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.758: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.758: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.758: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.758: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.758: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.758: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.758: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.762: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.762: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.762: invalid (NULL) pointer instance
[... 80 lines cut ...]

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.774: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.775: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.775: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): GLib-GObject-WARNING **: 17:04:56.775: invalid (NULL) pointer instance

(prusa-slicer:17966): GLib-GObject-CRITICAL **: 17:04:56.775: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:17966): Gdk-ERROR **: 17:05:07.907: The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadAlloc'.
  (Details: serial 28479 error_code 11 request_code 149 (unknown) minor_code 2)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
Trace/breakpoint trap (core dumped)
`

#### Project File (.3MF) where problem occurs
[issue.zip](https://github.com/prusa3d/PrusaSlicer/files/6912890/issue.zip)

combolek, 2021-08-02:
FWIW, I cannot reproduce it. It slices fine for me both with the official 2.3.1 AppImage or with the latest master compiled from source:
[image]
Did you try the AppImage?

Caraffa-git, 2021-08-02:
@combolek It turns out that the snap version has such a problem, I just tried AppImage and it's working correctly now. 
Thanks

closed 2021-08-02 by Caraffa-git