prusa3d/PrusaSlicer #6766: Prusa Slicer has double window borders on ElementaryOS

brodock, 2021-08-01:
### Version
2.3.3

### Operating system type + version

elementary OS 5.1.7 Hera
Linux 5.4.0-80-generic
GTK 3.22.30

### Behavior

you can see on the screenshot below that there are two windows decorations, both on the main window as well as on the about modal (and any other modal window):

[image]

### Proposed solution

This is a known issue with WxWidgets 3.1.0 which PrusaSlicer relies on as the GUI toolkit.

You can see the problem and solution being discussed here: https://trac.wxwidgets.org/ticket/17971

It seems Prusa Slicer already uses its own WxWidgets fork, so perhaps you should cherry pick the fix from the issue above.

bubnikv, 2021-08-02:
Thanks, we will consider merging the provided wxWidgets fix into our
wxWidgets 3.1.4 branch.

lukasmatena, 2021-08-04:
@brodock I may be wrong, but it seems that the commit you mention is already merged in our `v3.1.4-patched` branch (https://github.com/prusa3d/wxWidgets/tree/v3.1.4-patched) which we link to our binaries.

The title bar in your screenshot says 2.3.3+UNKNOWN. What wxWidgets version is that linked against? Does the issue also happen with Prusa-built AppImage that you can get from https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.3?

Thanks.

brodock, 2021-08-20:
I believe the flatpak was relying on system packages somehow. Since I upgraded to the new Elementary OS that doesn't happen anymore. I think we can close this

closed 2021-08-20 by brodock