prusa3d/PrusaSlicer #6769: Drag and drop of STL files stopped working after upgrade

russell-taylor, 2021-08-02:
### Version
2.3.3 (also happened on 2.3.1)

### Operating system type + version
Windows 10 21H1

### 3D printer brand / version + firmware version (if known)
Prusa MKS3+ and MMU2S

### Behavior
* When I drag am STL file onto the desktop icon, the program launches but does not load the file.  It does have a configuration update notification appear, not sure if that is relevant.  Previous versions on this same computer did open the STL files.  When I launch the STL file itself, Slicer does load it (and the configuration update notification still appears).

[image]

#### Project File (.3MF) where problem occurs
This happens with multiple .STL files, including files that loaded on previous versions.

bubnikv, 2021-08-02:
Is it possible that you installed some other slicer, which registered for
the STL or 3MF extensions?

russell-taylor, 2021-08-02:
When I open the file directly, it launches PrusaSlicer, so that part it working.  The problem happens when I drag the file onto the desktop shortcut, in which case PrusaSlicer is still run but it does not load the file.

FidelCapo, 2021-08-04:
I cannot reproduce this issue. Can you please attach video? Thanks

russell-taylor, 2021-08-05:
My display is two 4K monitors, so that would be a big video.  Here is a snip showing what file I drag to where to launch.
[image]
Here is a snip showing what it looks like after PrusaSlicer launches.
[image]

russell-taylor, 2021-08-05:
If I just click on the 3DBenchy.stl file and launch it, PrusaSlicer opens with the object loaded.
[image]

bubnikv, 2021-08-05:
Anybody else has this issue?

@russell-taylor 
Will it help, if you open the Configuration Wizard, go to the page "File association" and check the "associate 3MF" check box and Finish?

ssill2, 2021-08-05:
I just tested with 2.3.3  if you drag a .stl file to the 2.3.3 icon on the desktop, it opens the slicer, but nothing is imported.   To be honest I would probably never use PS in this way, but I can verify the behavior I saw.

ssill2, 2021-08-05:
I feel like what's probably missing is the short cut adding %*  so that the file names dropped on the icon are passed to the app.

ssill2, 2021-08-05:
That's not it.
I just tried this from command-line
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>"c:\Program Files\Prusa3D\PrusaSlicer\prusa-gcodeviewer.exe" "c:\Users\steph\Documents\modeling\0.2mm mouse ear.stl"

And PS launched without loading stl.

russell-taylor, 2021-08-05:
@bubnikv Setting that check box in the configuration wizard and clicking Finish did not help.  Of note, when I went back and tried to redo that after the program restarted the check box was still not checked.

fe60, 2021-08-17:
Drag and Drop of stl-files onto the shortcut of PrusaSlicer (2.3.3 on Windows 10 21H1) works for me with and without associating stl-files with PrusaSlicer (changed default app in Windows settings without using configuration wizard of PS).

PrusaViewer opens gcode files after dropping them onto the related application shortcut,

russell-taylor, 2021-08-22:
When I clicked on the STL file (which had an icon of the PrusaSlicer on it), Windows asked me which app to use to open it.  I selected PrusaSlicer.  It opened the program and opened the file.  Now, if I click on the STL file, things open as expected.

However, when I drag the STL file to the desktop shortcut, it still opens the app without the file open.

russell-taylor, 2021-08-22:
It looks like the desktop icon points to prusa-slicer updater rather than to the actually app, which must be where things are going sideways.

russell-taylor, 2021-08-22:
Yeah, if I drag the STL file onto the prusa-slicer.exe itself, everything works fine.  The updater is apparently not passing along the arguments to the program when it starts it, as @ssill2 suggested.

bubnikv, 2021-08-23:
@russell-taylor thanks for heads up. We will check with our college responsible for the Windows installer.

bubnikv, 2021-08-23:
It has been confirmed to me, that @russell-taylor 's findings are correct. Our new Windows installer installs a link to an auto-updater wrapper to PrusaSlicer. We will check with the vendor of the updater wrapper first whether they can help us.

bubnikv, 2021-12-04:
Fixed by starting the 3rd party updater application on Windows if it is available by PrusaSlicer, not vice versa.
This issue will be fixed for good with the Windows installer we will provide for PrusaSlicer 2.4.0 final.

9994e0bcbc14fdfbd46f57d3b04d5aad087f8c36
262304c777f4905fc74b9c4b5d1d9819d645d54d

Also for PrusaSlicer 2.5.0 we will likely implement our own solution for downloading application updates, which will be platform independent.

closed 2021-12-04 by bubnikv