prusa3d/PrusaSlicer #6771: SL1S File Extension

andrewluebke, 2021-08-02:
### Version
2.3.3

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa SL1S

### Behavior
When sending files to printer the default file extension is .SL1 which should be .SL1S.  PrusaSlicer should be able to figure out from the type of printer being used what extension to use automatically.

rtyr, 2021-08-03:
Are you sure you are using correct profiles? Please provide some 3mf project file where this problem occurs.

andrewluebke, 2021-08-03:
I just realized I can have a separate print settings profile for the SL1S vs the SL1.  However shouldn't the output options for the filename be under printer settings, since it's dependent on the printer type?

rtyr, 2021-08-03:
It doesn't have to be dependent on the printer type. Different printers can share same print profile and same output filename format in general (it is not the case with SL1/SL1S, because of different file extension). One can also use various variables in the filename (time estimate, name of the material, infill percentage (fdm), etc.). I guess you used some custom profile, because system SL1 print profiles are not available/visible for SL1S printer.

So I guess we can close this issue?

andrewluebke, 2021-08-03:
Sure, thanks

closed 2021-08-03 by rtyr