prusa3d/PrusaSlicer #6774: Floating point exception since commit 0d74502aeb84c4f8a9eebfea6d22b8abc71b6036

plampix, 2021-08-05:
### Version
master, ever since commit 0d74502aeb84c4f8a9eebfea6d22b8abc71b6036
Before this commit, this issue didn't exist

### Operating system type + version
Debian GNU/Linux 11

### 3D printer brand / version + firmware version (if known)
Not relevant

### Behavior
Starting prusa-slicer, even without a profile, crashes and displays the following message in the terminal:
floating point exception  prusa-slicer

#### Project File (.3MF) where problem occurs
Not relevant

abel5678, 2021-08-06:
Hay

kocikdav, 2021-08-09:
Hello, thank you for reporting. Unfortunately this crash does not happen on any of our Linux based computers. Can I ask you to run PrusaSlicer from terminal with --loglevel=5 ? We might see where exactly this problem occurs or at least see where to put additional debug logs.

lukasmatena, 2021-08-09:
@plampix You were able to tell which commit broke it, so I assume you are capable of building PrusaSlicer. Can you please build it in debug mode and tell us at which line exactly the problem occurs and what the backtrace looks like? Thanks.

plampix, 2021-08-09:
Sure, I'll try.

plampix, 2021-08-09:
I've found the problem!

prusa-slicer tries to read /usr/local/share/PrusaSlicer/data/hints.ini, which isn't installed by the 'make install'. Copying the file manually solves the problem.

lukasmatena, 2021-08-09:
@plampix Thanks a lot. Do I guess right that you are using SLIC3R_FHS? The problem would then be caused by https://github.com/prusa3d/PrusaSlicer/commit/454b8764203ed3601945e656c2c122a361b8af7c, which EXCLUDEd data subdirectory from install target. We should fix this.

Also, we should definitely improve error reporting.

plampix, 2021-08-09:
@lukasmatena Yes, I do use SLIC3R_FHS.

kocikdav, 2021-08-09:
Thanks, my commit should prevent the crash. Meanwhile deletion should be solved in #5930.

closed 2021-08-09 by kocikdav

lukasmatena, 2021-08-12:
According to https://github.com/prusa3d/PrusaSlicer/commit/454b8764203ed3601945e656c2c122a361b8af7c#comments and https://github.com/prusa3d/PrusaSlicer/commit/3b70d329ba474e01e79fe234a7ec7a70f2d9af02#comments, commit  52718e96f5de49162c17f3b0335e982a2580c73e should fix this. Thanks @plampix , @dimitry-ishenko and @xarbit.

Closing.