prusa3d/PrusaSlicer #6775: Filaments are not available for all printers

davidewen, 2021-08-05:
### Version
2.3.3+arm64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
MacOS 11.4 Big Sur
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Prusa MK3s
Creality Ender 6

### Behavior
filaments are not available for all printers

rtyr, 2021-08-06:
Intended behavior. Filament profiles are compatible only with printers in its vendor space and can be also further restricted to specific printer models. Closing.

closed 2021-08-06 by rtyr

davidewen, 2021-08-06:
How do I make them compatible?

ssill2, 2021-08-06:
in the filament profile there's some code that makes that determination.   You can make a copy of the filament profile and then adjust that code to make it show up for your printer.

davidewen, 2021-08-06:
Ok. 
It seems strange to me that you have check boxes for “ALL” printers and “ALL” profiles, but yet you still have change code?

I’m assuming you are referring to this code?

ssill2, 2021-08-06:
This code
[image]

ssill2, 2021-08-06:
This is the same stuff that prevents FLEX filament from being available for MMU printers.

davidewen, 2021-08-06:
We may not be talking about the same thing.

I have a MK3s and an Ender 6

All of my filament types are available to the MK3s, but only 3 generic ones are available to my Ender 6.

ssill2, 2021-08-06:
I would still take a look at this stuff for the ones that aren't showing up for your printer.  I know I have copied FLEX filament profiles and edited so that they would show up when the MMU was selected.

davidewen, 2021-08-06:
What do I need to modify/add/remove to make it show up?

What happens if I “detach from system preset” in “Dependencies” ?

It sure seems like this could be a lot easier….
There seems to be multiple locations to make changes, with no explanation or instructions of what does what and for what reason.

David

ssill2, 2021-08-06:
I only have prusa printers, so I'd have to let someone with more experience than I answer that one.

rtyr, 2021-08-06:
You can create a copy of the system profile meant for Prusa printer and make it visible for all printers. 

1. Select the profile you want
2. Click on "Detach from system preset" (red arrow)
3. There will be message informing you about creating a copy of the profile.
4. Modify or remove all from "Compatible printers condition" (blue arrow) and Save.

This new profile will be visible for all printers. It may not work properly with your Ender without modifications though. (temperatures, linear advance values, retraction overrides, volumetric speed limits, etc. ...).

There is also an option Preferences -> Show incompatible print and filament presets. It will enable special flag, which will let you toggle incompatible profiles on and off (orange arrow).

[image]

rtyr, 2021-08-06:
Also please note that filament profiles for 3rd party vendors (like Creality) are added based on the community contributions.