prusa3d/PrusaSlicer #6777: Bridging collides with the speed of the small perimeter radius

Henyckma-Sensei, 2021-08-06:
### Version

2.3.3 + Win64

### Operating system type + version
Windows 10 x64bits

### 3D printer brand / version + firmware version (if known)
Custom prusa i3 with Marlin 2.0.7

### Behavior
When presenting a bridging, the program takes as priority the speed of the small perimeters causing the material to fall down without taking into account the speed of the bridging.

[image]

rtyr, 2021-08-07:
Please provide your 3MF project file.

Henyckma-Sensei, 2021-08-07:
> Please provide your 3MF project file.

[GPU_Bracket_Rev6.zip](https://github.com/prusa3d/PrusaSlicer/files/6949394/GPU_Bracket_Rev6.zip)

Henyckma-Sensei, 2021-08-13:
> Please provide your 3MF project file from PrusaSlicer (it contains all settings etc.). File -> Save project as...
> The 3MF you provided seems to be created in MS 3Dbuilder.

[GPU_Bracket_Rev.zip](https://github.com/prusa3d/PrusaSlicer/files/6984260/GPU_Bracket_Rev.zip)

sorry i'm new here

rtyr, 2021-11-09:
I was able to reproduce it on current master.

lukasmatena, 2021-11-10:
@FidelCapo Just FYI. There is a PR addressing this issue: #6044.

closed 2026-09-29 by prusaslicer-bot[bot]