prusa3d/PrusaSlicer #6781: Head moves too high (z-axis) after completing small print on Creality printers

adam-ah, 2021-08-08:
### Version
2.2.3

### Operating system type + version
Applies to all versions

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 v2

### Behavior
After completing small prints, the head moves unnecessarily high (0.6 of max height).

#### _Steps needed to reproduce the problem_

Print anything tiny.

#### _Expected Results_

The head should move slightly above the highest layer, but not too much above.

#### _Is this a new feature request?_

No

### Further details

In the [Creality.ini](https://github.com/prusa3d/PrusaSlicer/blob/master/resources/profiles/Creality.ini#L877) file, the final height setting is not taking into account the height of the print:

```
[ln01] {if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+2, max_print_height)} F600 ; Move print head up{endif}
[ln02] G1 X5 Y{print_bed_max[1]*0.8} F{travel_speed*60} ; present print
[ln03] {if max_layer_z < max_print_height-10}G1 Z{z_offset+min(max_layer_z+70, max_print_height-10)} F600 ; Move print head further up{endif}
[ln04] {if max_layer_z < max_print_height*0.6}G1 Z{max_print_height*0.6} F600 ; Move print head further up{endif}
[ln05] M140 S0 ; turn off heatbed
[ln06] M104 S0 ; turn off temperature
[ln07] M107 ; turn off fan
[ln08] M84 X Y E ; disable motors
```

The last move command (ln04) moves the head to 60% of the max height, irrespective of the `max_layer_z`, while the first move commands (ln01 and ln03) take into account the `max_layer_z`

### Resolution

Remove the last two unnecessary move commands (ln03 and ln04), the first one will be enough.

rtyr, 2021-08-08:
@pmjdebruijn fyi

pmjdebruijn, 2021-08-08:
If you check the end g-code you would see that the last move is executed specifically for low prints.

I've added both so you have you some extra room to reach and get the print off the bed, without having the print head be in the way. How helpful this is depends on the specifics of the print and the type of bed.

But in the worst case scenario there is no significant downside really.

You're of course free to customize your own install to your own personal preferences.

@rtyr as far as I'm concorned this can be closed as INVALID.

adam-ah, 2021-08-08:
I'm missing something, @pmjdebruijn: what's the validity of using 3 Z move codes, where the last one has no minimum height, and one move code would be sufficient?

> But in the worst case scenario there is no significant downside really.

There is a huge downside: Z movements are slow, so when printing small parts, users not only need to wait for the printer to complete this unreasonably long move operation, but they will also need to wait an excessive amount of time for the head to come down for the next print. 

Sending the head up just so we can wait for it to come down is nothing but a waste of time.

> You're of course free to customize your own install to your own personal preferences.

As it's the default behaviour, I must ask: would more people benefit from this feature than not? And how easy it is for beginners to change this behaviour?

I have not heard anyone asking for such a feature, and other slicers (e.g., Cura) do not have similar behaviours either.

I am struggling to see this as a _feature_; it is rather a _bug_, for the reasons above (3 moves when 1 would be enough; time takes to move up then down for the next print).

pmjdebruijn, 2021-08-08:
The end_gcode is shared across all Creality printers (with differing printer dimensions and different bed types), therefore it's a three stage process:

First we move 2mm in Z, to clear the print,and move the print head out of the way.
Then we move up an additional 70mm to allow for move space around the print for easy removal
For relatively low prints, we move to 0.6*height, if you're not already past that point, again for easy removal
I added this last part, because I noticed that the lower the print, the more the clearance matters for convenience

It's three commands because they depend on different conditions. It could probably be collapsed into a single move, using some min()/max() and some math, but the g-code would not be as clear in intent as it is now, and I highly prefer the readability over having it be a single command.

@waste of time, it's just a few seconds... And I don't see compromising easy print removal for a tiny gain in time as reasonable, at least not as far as default behavior for the general public goes.

@modify, I said you can modify if you want, most people will simply not care... So it doesn't have to be easy to modify at all, given the behavior is fine for most people anyhow.

@cura, I don't see how Cura's behavior matters one way or the other...

adam-ah, 2021-08-08:
Thanks for the detailed response, you highlighted some important problems. Let's focus on these:

1.
>  And I don't see compromising easy print removal for a tiny gain in time as reasonable, at least not as far as default behavior for the general public goes.

I'm still a little bit lost - did the _general public_ ask for this feature, or you decided it would be a good idea to include this? Was there a discussion about this feature or it was just added? I could not find an issue/discussion tagged with this commit.

2.
 I can see that [your change was added as  _"Various improvements, see commits."_](https://github.com/prusa3d/PrusaSlicer/pull/5863). 

Maybe you do see this as an improvement, but others may not (hence the report). 
There is no need to provide more evidence for a reversal of feature than there was evidence supplied in the first place when it was added (none, in this case).

As I can see, no one asked for this feature and you did not provide evidence to support its usefulness. 

On the other hand, I provided evidence that it is detrimental (e.g., waste of time between prints; atypical behaviour compared to other slicer software; unnecessary lines of GCODE; unnecessary wear&tear on the stepper and rod).

2. 
As for easy removal, I don't see how this make it any easier at all - the head is not in the way for low prints: low prints tend to be narrow as well (i.e., generally small, not just low), and therefore they are sitting on the middle of the build plate, and easy to remove them.

Could you perhaps come up with an example that happens more often than not and this move would help (I cannot think of any)?

3. 
> I said you can modify if you want

Maybe the customisation idea goes both ways: if you want to move _your_ printer head all the way up after tiny prints, _you_ can customise your own settings, rather than changing every Creality printer users' settings.
This is especially important as there was no issue reported, nor a discussion about what problems this change aimed to solve.

Thanks for your time.

pmjdebruijn, 2021-08-09:
Re 1: I made the change without there being a user request for it. Creality.ini was essentially unmaintained, so for most practical purposes I sortof kindof took over maintainership, since nobody else did for quite a while.

Re 2: This is were you are quite definitively wrong. The current state, is as it is, any change needs to be argued for. If there isn't a compelling argument for change, things stay the way they are. And your arguments are either tied to very specific use-cases (tiny prints), or are really thin (minor added wait, tiny amount of wear).

Re 3: See Re2: given I feel the current defaults are probably better for more people. Obviously there is no objective way to prove this one of or the other. But...

This has been the way it is for roughly 7 months now, and in that time, we've had one other ticket from someone who asked as well about the three moves thing. But you're the first to make a ticket about the extra wait time in all that time, and given how common Creality printers are, that sortof would somewhat indicate the current state of affairs probably isn't that poor really...

@rtyr, unless you strongly disagree, I have no intent to change anything based on the points made in this ticket, so it can be closed.

As a sidenote, in the future I would be open to retuning the 0.6 multiplier in the last move, as this may be a little much for larger printers like the CR-10 and friends, but I'm not sure yet how to approach that in a way that works for all printers.

adam-ah, 2021-08-09:
> The current state, is as it is, any change needs to be argued for. 

You did not put anything forward when you added the change; on the other hand, you say that I have to bring evidence why this change is better to be removed? It's a case for [Hitchens's razor](https://en.wikipedia.org/wiki/Hitchens%27s_razor)

> given I feel the current defaults are probably better for more people

Facts over feelings: I added multiple examples of why the change is not desirable, but you did not add any examples to demonstrate when it's needed.

> I have no intent to change anything

I think you might be confused here: no one asked you to change anything; put differently, the codebase is not yours.
@rtyr simply asked you to add your thoughts, as you made the change recently. So far, you have not added any logic-based reason why this change is desirable.

pmjdebruijn, 2021-08-09:
> You did not put anything forward when you added the change; on the other hand, you say that I have to bring evidence why this change is better to be removed? It's a case for [Hitchens's [...]

The argument for is easier print removal. For example earlier Creality printers did not come with a removable print bed, so having access to a print from all sides is helpful in a such a case. But even with a removable bed, having easy access to a print is a nice to have. Quite often I don't have to take my bed off at all.

> > given I feel the current defaults are probably better for more people
> 
> Facts over feelings: I added multiple examples of why the change is not desirable, but you did not add any examples to demonstrate when it's needed.

Neither are really fact based though. Ultimately it's just what I think is best vs what you think is best.

> I think you might be confused here: no one asked you to change anything; put differently, the codebase is not yours.
> @rtyr simply asked you to add your thoughts, as you made the change recently. So [...]

@rtyr notified me of this ticket, since I've been the de-facto maintainer of Creality.ini for almost a year now, as I sortof mentioned previously. While ultimately @rtyr makes all the final decisions as he has commit access, however most of the time he defers to my judgement when it concerns Creality.ini

sth519, 2021-08-09:
If I may add something to this discussion: The fact that PrusaSlicer does this by default was one of the many little things that made me prefer it over Cura. With Cura, I ended up manually raising the Z-Axis manually after prints more often than not. Even just for being able to conveniently clean the print bed before the next print. And doing it manually takes a lot longer (going to the menu, pressing up multiple times).

I don't really get the reasons presented here for why raising the Z axis a bit is a bad thing. It literally takes seconds on my machine and I'm definitely not worried about additional wear from that little bit of movement or my files being a few bytes larger.

rtyr, 2021-08-09:
Since it's implemented for quite some time without complaints, it seems most users are fine with it. There will always be different opinions about various things in system profiles. As mentioned, @pmjdebruijn maintains Creality bundles for long time. I think we can close this issue.

closed 2021-08-09 by rtyr