prusa3d/PrusaSlicer #6782: Clicking On Another Modifier while Changing a Modifier Parameter crashes PrusaSlicer

JackWitherell, 2021-08-08:
### Version
2.3.2+win64

### Operating system type + version
Windows 10 19042

### 3D printer brand / version + firmware version (if known)
MK3S+

### Behavior
When changing a modifier parameter, if you don't click away of press enter and instead *click on another modifier* then PrusaSlicer quietly crashes and quits out, dumping all your changes.

Steps to reproduce:
  Open PrusaSlicer
  Create a new object (right click new part works)
  Create a modifier on that object (box modifier works fine)
  Create another modifier (duplicating the first one works fine)
  Right-click either modifier and set layer parameters
  Type a number in any of the three value slots at the bottom right
  **Click on the other modifier in the 3D View**

Expected Results
  Slicer Applies the changed parameter and moves over to the newly selected modifier

Actual Results
  Entire program crashes (quits out silently)

#### Project File (.3MF) where problem occurs
N/A

fe60, 2021-08-09:
Changing parameters is not required. It crashes, as soon as the parameters are visible while right-clicking the other modifier.
* Open [PrusaSlicerIssue6782.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6952946/PrusaSlicerIssue6782.3mf.zip) (just remove file ending `.zip`),
* select node "Layers and Perimeters" of modifier A (right) and
* perform a right click on modifier B (left).
=> Crash

[image]

lukasmatena, 2021-08-09:
Thanks, I managed to reproduce the crash in 2.3.0, 2.3.2 and current master. It seems to be ok in 2.2.0. The **right** click is the key.
<details>
<summary><b>Debug mode backtrace attached.</b></summary>

1   __GI_raise                                                                                                                                                                                                                                                   raise.c            54   0x7ffff479d438 
2   __GI_abort                                                                                                                                                                                                                                                   abort.c            89   0x7ffff479f03a 
3   __assert_fail_base                                                                                                                                                                                                                                           assert.c           92   0x7ffff4795be7 
4   __GI___assert_fail                                                                                                                                                                                                                                           assert.c           101  0x7ffff4795c92 
5   Slic3r::GUI::ObjectList::get_selection_indexes                                                                                                                                                                                                               GUI_ObjectList.cpp 350  0x11951ac      
6   Slic3r::GUI::MenuFactory::append_menu_items_convert_unit                                                                                                                                                                                                     GUI_Factories.cpp  763  0x1178108      
7   Slic3r::GUI::MenuFactory::part_menu                                                                                                                                                                                                                          GUI_Factories.cpp  999  0x117a273      
8   Slic3r::GUI::Plater::priv::on_right_click                                                                                                                                                                                                                    Plater.cpp         3928 0x10fefae      
9   wxEventFunctorMethod<wxEventTypeTag<Slic3r::GUI::Event<std::pair<Eigen::Matrix<double, 2, 1, 2, 2, 1>, bool>>>, Slic3r::GUI::Plater::priv, Slic3r::GUI::Event<std::pair<Eigen::Matrix<double, 2, 1, 2, 2, 1>, bool>>, Slic3r::GUI::Plater::priv>::operator() event.h            425  0x114e1f7      
10  wxAppConsoleBase::CallEventHandler                                                                                                                                                                                                                           appbase.cpp        671  0x25227bf      
11  wxEvtHandler::ProcessEventIfMatchesId                                                                                                                                                                                                                        event.cpp          1425 0x25c6136      
12  wxEvtHandler::SearchDynamicEventTable                                                                                                                                                                                                                        event.cpp          1897 0x25c6e03      
13  wxEvtHandler::TryHereOnly                                                                                                                                                                                                                                    event.cpp          1618 0x25c657e      
14  wxEvtHandler::TryBeforeAndHere                                                                                                                                                                                                                               event.h            3927 0x25c790d      
15  wxEvtHandler::ProcessEventLocally                                                                                                                                                                                                                            event.cpp          1555 0x25c63cd      
16  wxEvtHandler::ProcessEvent                                                                                                                                                                                                                                   event.cpp          1528 0x25c6364      
17  wxEvtHandler::ProcessPendingEvents                                                                                                                                                                                                                           event.cpp          1392 0x25c6019      
18  wxAppConsoleBase::ProcessPendingEvents                                                                                                                                                                                                             [... cut ...]
20  wxapp_idle_callback                                                                                                                                                                                                                                          app.cpp            99   0x242e0d6      
... <More>                                                                                                                                                                                                                                                                                              

</details>

It asserts on `assert(!sels.IsEmpty())`. Probably something for @YuSanka.

bubnikv, 2021-08-11:
Fixed in PrusaSlicer 2.4.0-alpha1.

closed 2021-08-11 by bubnikv