prusa3d/PrusaSlicer #6783: idea: easier material/color change

tkapler, 2021-08-09:
it would be nice (option) to automaticaly create someting like a desk modificator (but full width/length), when i insert color change and allow filament change for it. So that if i got e.g. two brands of PLA and i want to do two color object, or even two types of filament, i could easily switch it at the color defined height

ssill2, 2021-08-09:
I'm having trouble visualizing what you're asking for that's not an MMU.

rtyr, 2021-08-10:
I am afraid I don't understand your request.

tkapler, 2021-08-11:
Imagine you want to print dual color object with simple manual color swap - so e.g. first white desk and then black text on it. You do it on Preview window with adding the M600 color change command. Great.
But now imagine, that you want to print first color with Prusament PLA and the second with generic PLA (or worse - first with e.g. PLA and second with ABS). There is no easy way (or I do not know it) how to say - print this color with this filament and the second with another one. You can somehow solve it by adding a modificator and then modify it manualy for every setting that is different for the second filament, but that would be realy pain. But it could be easily automated.

ssill2, 2021-08-11:
ok that makes more sense.  The way you were describing it sound sounded like parts of the same layer would be printed in different colors.

rtyr, 2021-08-11:
So basically https://github.com/prusa3d/PrusaSlicer/issues/2002, https://github.com/prusa3d/PrusaSlicer/issues/2469, https://github.com/prusa3d/PrusaSlicer/issues/2826, https://github.com/prusa3d/PrusaSlicer/issues/3378.

rtyr, 2021-08-13:
Closing as a duplicate.

closed 2021-08-13 by rtyr