prusa3d/PrusaSlicer #6785: Suggestion: Select multiple profiles at once

BrianKramp, 2021-08-09:
### Version
2.3.1

### Operating system type + version
Windows 10 latest

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S+ latest firmware

### Behavior
_Is this a new feature request?_ YES

I want to have a base profile that I can almost always apply, and then I want to be able to save little tweaks on top of it. But I can only save and apply an entire profile. I want something that will allow me to select a base profile, and then have an override, or patch profile that only has a few settings in it.

For example one of my presets is to have a slower speed than others. But I don't want to save an entirely new settings file for just the slow speed. I want to apply:

PLA Base Profile
Slow Speed Override Profile // Just defines new speeds

Other times I want extra params and infill, so I would apply:

PLA Base Profile
Extra Strength Override Profile / / Just defines new perims and infill

I could easily have 5-6 override profiles that I apply and remove. In the current system, I'd have to save 5-6 profiles, but each time I learn some new benefit that I should apply everywhere, I have to apply it to every profile, and that would be too burdensome and error prone.

Theoretically, then I could ask for any number of overrides:

PLA Base Profile
Slow Speed Override Profile
Extra Strength Override Profile

I like this because it seems like you could release a few of these override profiles, and have an even more beginner-friendly software, and yet advanced users will also love it.

neophyl, 2021-08-10:
You need to look into profile dependencies.  That's pretty much how they work.  Settings are stored in a profile and other profiles use those base settings and any settings in that profile replace the same one in the lower one its built upon.  The only problem is that at the moment there is no editor for this functionality so you have to manually do so.

One of the Prusa forum users has a website with some useful info on this that is worth a read https://projects.ttlexceeded.com/3dprinting_prusaslicer_profiles.html

BrianKramp, 2021-08-12:
Looks like the underlying system is already in place then, we just need a way to expose it in the UI. That's promising. I really think this could make slicing more approachable to users by having simple built-in overrides like "increased bed adhesion" instead of tinkering with 5 values for skirts, brims and rafts.

closed 2026-09-29 by prusaslicer-bot[bot]