prusa3d/PrusaSlicer #6786: PrusaSlicer incorrectly sorts print profiles (by number)

FIRES-3D, 2021-08-09:
Version: PrusaSlicer V2.3.3+win64
OS: Windows 10 Pro 64-bit
Printer: Prusa MK3S

Behavior:
When selecting between print profiles, PrusaSlicer does not correctly sort between names that include numerical values. PrusaSlicer sorts based on leading numerical characters rather than the entire number. This makes it far more annoying to keep profiles organized and navigate between them (especially when you have many to work with!)

CURRENT BEHAVIOR: Given the example profile names below, PrusaSlicer sorts in the following order:
N0.4 S240 H0.25
N0.4 S45 H0.25
N0.4 S90 H0.25

EXPECTED / DESIRED BEHAVIOR: PrusaSlicer should sort in the following order:
N0.4 S45 H0.25
N0.4 S90 H0.25
N0.4 S240 H0.25

Here, the value "S240" should be sorted after "S90", not before "S45".

neophyl, 2021-08-10:
That's fairly normal for file sorting on a lot of computers though.  Its not parsing as a human where it looks and interprets the string, its simply matching on next character.  If you want 45 to be before 240 then you need to use 045 in the name.
Our systems at work do the same and I don't think its a trivial change to alter it as it tends to rely on the built in functions to work and that's your OS.

Maybe they can do something about it though depending on what library is in use.

bubnikv, 2021-08-10:
As noted by @neophyl, profiles are sorted alphabetically, not numerically.

FIRES-3D, 2021-08-12:
Yep, I understand "why" it sorts this way, and that there are workarounds. However I'm pointing this out because for many users this is not a desirable or intuitive behavior. Improving the sorting to include numeric sorting would improve QOL as well as profile readability (by eliminating the need for leading 0's)

murk-sy, 2021-08-13:
You can substitute a leading zero with an underscore which improves readability. What you're proposing would be some sort of number-aware alphabetic sorting, which would be the definition of unintuitive behaviour. Just because you use temperature (or whatever that number actually is) in your profile names, that doesn't mean everyone or frankly anyone does. Some use model numbers (of the print) with letters in them, just for an example. Differentiating between what is a "real" number and what is just a code is not generically possible.

You might have noticed that the default profiles have layer heights listed in milimeters, despite none of them ever being even one milimeter. While consistency with the rest of the program is the main reason for that, another reason is that is basically forces usage of leading zeroes, thus proper and logical sorting.

FIRES-3D, 2021-08-13:
I'm unsure why you would say this is "the definition of unintuitive behavior" considering many programs (including OS file explorers) use number-aware alphabetic sorting by default. For many users (myself included), number-aware sorting is the more intuitive method given its current prevalence.

I understand we likely just disagree on preference here. However this is a suggestions forum and my purpose is simply to highlight a point of friction I've had with the software as a Prusa customer. I respect that there are logical reasons for using the current implementation, but in the interest of potentially improving QOL for other users that might be facing the same irritation (and a behavior that is unlike that in many other programs / file explorers), I am voicing my observation here. Just trying to help.

closed 2026-09-29 by prusaslicer-bot[bot]