prusa3d/PrusaSlicer #6787: Add "update STLs" to project

jkuusama, 2021-08-10:
### Version
Version 2.3.3+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Unrelevant (but MK3S+)

### Behavior
Scenario: A design project has many parts. These are imported to slicer and saved as a project. First print for a prototype didn't work as desired. The design is improved. It would be very helpful if all the STLs of the parts in a project could be updated to the latest versions with one operation.

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
n.a.

lukasmatena, 2021-08-10:
I think you have just described **Reload from disk** function, which is already implemented (`Edit->Reload from disk`). Is that not enough for your needs for some reason?

jkuusama, 2021-08-10:
Hi Lukáš,

Hmm... Yes, I did. I believe I tried that earlier today, and I was asked for a file name of one of the parts. I tried that again a moment ago, and indeed, it works just as I wanted. I have no idea about why it didn’t work in the morning, but that doesn’t matter. Sorry to bother you!

Best regards

Juha
--
Juha Kuusama
LitePlacer Inc
Palomäentie 15 B
33230 Tampere, Finland
tel +358 40 705 8815
***@***.******@***.***>

lukasmatena, 2021-08-10:
@jkuusama No bother. The reason you were asked is because slicer could not find the original file. It may be connected to `Export full pathnames of models and parts sources into 3mf and amf files` option (available in Preferences), which is disabled by default and was implemented for security reasons when such a 3mf would be shared etc.

Closing.

closed 2021-08-10 by lukasmatena

jkuusama, 2021-08-10:
Hi,

Thank you! That was indeed the issue. Thank you for your support!

Best regards

Juha