prusa3d/PrusaSlicer #6790: Different starting point of inner and outer wall of perimeter

richardklingler, 2021-08-10:
### Version
Version 2.3.3+linux-x64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Debian 10.10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Ender-5 Plus, Marlin 2.0.9.1

### Behavior
* _Describe the problem_
The inner and outer wall of a perimeter don't have the same starting point causing defects on the outside of the printed wall.
This happens with start set to "Rear", "Aligned" and "Nearest".

[image]

[image]

* _Steps needed to reproduce the problem_
Download the Benchy STL and slice it with 0.2mm height and starting point set to "Aligned".
At the second layer (0.4mm) it can be seen that the inner wall starts at the front of the boat, whereas the outer wall starts at the back.

* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[benchy.zip](https://github.com/prusa3d/PrusaSlicer/files/6961024/benchy.zip)

richardklingler, 2021-08-10:
BTW: Slic3r 1.3.1-dev shows the correct behavior with the same print settings....start point of inner and outer walls are the same...

bubnikv, 2021-08-10:
I suppose it is due to the Monotonic infill, which always fills from one
side to the other. If you disable monotonic infill for the top / bottom
surfaces, you will get the Slic3r 1.3.1-dev behavior.

richardklingler, 2021-08-10:
Makes no difference when setting top/bottom to rectangular infill...

Inner wall starts at front of the benchy...and outer wall at the back....

murk-sy, 2021-08-24:
Are you perhaps using seam painting? This is an issue with seam painting (which is somewhat frustrating to me)

See also #6400 (I'm collecting similar seam issues there)