prusa3d/PrusaSlicer #6792: Large Gaps In Seams

Mustered, 2021-08-10:
### Version
2.3.3

### Operating system type + version
Windows 10 4 bit

### 3D printer brand / version + firmware version (if known)
Stock Original Prusa Mini, latest firmware.

### Behavior
First, let me apologize for doing this wrong if I am. I have requested help in multiple groups and forums to no avail and was sent here. I am not a coder, nor particularly savvy with this stuff. I am going to be detailed:

Since February 2020 I have been printing cookie cutters with walls .5mm/1mm thin with my .4mm nozzle. They printed perfectly. In April of 2021, so a full year after printing this way, I began experiencing these gaps in my seams. They started small and sporadic, and now are large, as pictured. The occurrence has increased until I'm now tossing about 2/3rds of my prints due to this issue. I sell cutters, so this is a real problem. 

The issue is not always repeatable. It used to be I could print a file, have these gaps, reprint and it was fine. Now that's not the case. This happens on seams in corners, on curves, and on straight edges. 
Print thin walls is checked, and unchecking created a cutting wall that was much too thick for clean cuts. 

So far I have:
Changed nozzles twice. Checked for clogs 100,000 times.
Adjusted retraction and detraction speed and amounts.
Changed where my seams were placed. "Random" just left smaller versions of these holes all over. "Rear", "Nearest," and "Aligned" just puts these holes in a big line.
Raised and lowered temps. I've tried increments from 190 to 225.
Increased and decreased the extrusion multiplier in increments from 1.0 to .9 and 1.1
Increased and decreased all speeds.
Decreased perimeters on the thinnest layer to 1.
Increased and decreased layer height.
Changed wipe settings, turned on and off retraction.
Printed with a more common .6/1.2 mm wall (many cutter makers use this thickness), had the same issue. I'm not willing to go any thicker.
Tried different detail presets.

Probably more. I've done these alone and in combination with other changes, but I'm game to try any combination you have in mind. I've Googled and Googled, and the few things I found discussing my problem didn't help. I've email Prusa, but it was only this morning. 
Where do I go from here?

_Is this a new feature request?_
I don't think so.

#### Project File (.3MF) where problem occurs
[image]

rtyr, 2021-08-11:
Based on your description (gradual change), I don't think the issue is PrusaSlicer related. Please contact our tech-support via live chat at https://shop.prusa3d.com or by e-mail. I would start by replacing the nozzle and short PTFE tube inside the hotend, then check whole filament path (extruder, long PTFE tube, PTFE couplers,...). Tech-support guys will help you with troubleshooting.

That being said, if you provide some example 3MF project file, we can try to print it.

bubnikv, 2021-09-22:
With single line walls, you may also disable extruder wipes.