prusa3d/PrusaSlicer #6794: 2.4.0-Alpha0 crash when adding brim while in preview

jmd73, 2021-08-11:
### Version
Built from source
version_2.4.0-alpha0-1030-g4c646905f
4c646905fd462b71a74754030ff2dc668b7880b9

### Operating system type + version
Windows 10 Pro 64bit

### 3D printer brand / version + firmware version (if known)
MK2.5S, MK3S + MMU2S

### Behavior
Crashes to desktop when changing any settings while in preview. (No, not just the brim setting but it was easy to change.)

* _Steps needed to reproduce the problem_
Start PrusaSlicer.
Add a box, for example.
Let it slice (I have background slicing active). (It will crash if switching before slicing is finished.)
Switch to Preview, if it isn't shown automatically.
Activate brim in quick-settings.
After a short while, crash.

* _Expected Results_
Slices without crashing.

* _Actual Results_
Crash at GLGizmosCommon.cpp row 75 in function CommonGizmosDataPool::instances_hider()
InstancesHider* inst_hider = dynamic_cast<InstancesHider*>(m_data.at(CommonGizmosDataID::InstancesHider).get());
And the reason seems to be because m_data is a <Struct at NULL> as Visual Studio shows it.

The function is called from GLCanvas3D.cpp line 5110, in function _render_objects.

Unfortunately I don't have time to dig in figure out a fix (I don't know the code good enough.)
If more data is needed or I have a bad build because something is not up to date. Please let me know.

lukasmatena, 2021-08-11:
@jmd73 Thanks. Could you please verify that https://github.com/prusa3d/PrusaSlicer/commit/7f2ad140b63711e0bf4db6cbf19fc0a60983f296 fixed it?

jmd73, 2021-08-12:
@lukasmatena Yes, this fixes the crashing!

Great work!

I'll close this issue.

closed 2021-08-12 by jmd73