prusa3d/PrusaSlicer #6796: 2.4.0-Alpha0 crash when select Thick Bridges

justoeu, 2021-08-12:
### Version
Built from source
version_2.4.0-alpha0-1016-g2ab0ea4c4

### Operating system type + version
MacOS BigSur 11.4 - Apple M1

### 3D printer brand / version + firmware version (if known)
MK2.5S, MK3S + MMU2S

### Behavior
Crashes to desktop when use the feature "Thick Bridges" selected. This behavior occurs when a model sliced.

- Steps needed to reproduce the problem
Start PrusaSlicer.
Include the Model, in this case, using drag & dropped;
inside print Settings config, select the "Thick Bridges"
Let it slice and few seconds, Crash.

- Expected Results
Slices without crashing.

- Actual Results
Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread
Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000058
Exception Note:        EXC_CORPSE_NOTIFY
Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [99191]
VM Regions Near 0x58:
--> 
    __TEXT                      1002a8000-101b60000    [ 24.7M] r-x/r-x SM=COW  /Users/*
Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.prusa3d.slic3r/           	0x00000001008961ec Slic3r::GLVolume::SinkingContours::update() + 356
1   com.prusa3d.slic3r/           	0x00000001008961d8 Slic3r::GLVolume::SinkingContours::update() + 336
2   com.prusa3d.slic3r/           	0x000000010089ac4c Slic3r::GLVolumeCollection::render(Slic3r::GLVolumeCollection::ERenderType, bool, Eigen::Transform<double, 3, 2, 2> const&, std::__1::function<bool (Slic3r::GLVolume const&)>) const + 572
3   com.prusa3d.slic3r/           	0x00000001008b01f4 Slic3r::GUI::GLCanvas3D::_render_objects(Slic3r::GLVolumeCollection::ERenderType) + 1028
4   com.prusa3d.slic3r/           	0x00000001008ae8a0 Slic3r::GUI::GLCanvas3D::render() + 804
5   com.prusa3d.slic3r/           	0x00000001008b9288 Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&) + 348
6   com.prusa3d.slic3r/           	0x00000001015fdc7c wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const + 200 (appbase.cpp:671)
7   com.prusa3d.slic3r/           	0x00000001016daee8 wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) + 236 (event.cpp:1425)
8   com.prusa3d.slic3r/           	0x00000001016dc7c8 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 224 (event.cpp:1897)
9   com.prusa3d.slic3r/           	0x00000001016dc5f8 wxEvtHandler::TryHereOnly(wxEvent&) + 80 (event.cpp:1618)
10  com.prusa3d.slic3r/           	0x00000001016dc378 wxEvtHandler::TryBeforeAndHere(wxEvent&) + 72 (event.h:3927)
11  com.prusa3d.slic3r/           	0x00000001016dc3b8 wxEvtHandler::ProcessEventLocally(wxEvent&) + 40 (event.cpp:1555)
12  com.prusa3d.slic3r/           	0x00000001016dc23c wxEvtHandler::ProcessEvent(wxEvent&) + 200 (event.cpp:1528)
13  com.prusa3d.slic3r/           	0x00000001016dc8fc wxEvtHandler::SafelyProcessEvent(wxEvent&) + 44 (event.cpp:1646)
14  com.prusa3d.slic3r/           	0x0000000101468cb0 wxWindowBase::HandleWindowEvent(wxEvent&) const + 36 (wincmn.cpp:1570)
15  com.prusa3d.slic3r/           	0x000000010146f018 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 104 (wincmn.cpp:2864)
16  com.prusa3d.slic3r/           	0x000000010146f088 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 216 (wincmn.cpp:2873)
17  com.prusa3d.slic3r/           	0x000000010146f088 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 216 (wincmn.cpp:2873)
18  com.prusa3d.slic3r/           	0x000000010146f088 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 216 (wincmn.cpp:2873)
19  com.prusa3d.slic3r/           	0x000000010146f088 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 216 (wincmn.cpp:2873)
20  com.prusa3d.slic3r/           	0x0000000101352c64 wxAppBase::ProcessIdle() + 200 (appcmn.cpp:406)
21  com.prusa3d.slic3r/           	0x0000000101553410 wxApp::ProcessIdle() + 48 (app.cpp:358)
22  com.prusa3d.slic3r/           	0x00000001016309ec wxEventLoopBase::ProcessIdle() + 48 (evtloopcmn.cpp:107)
23  com.prusa3d.slic3r/           	0x00000001016eb74c wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 124 (evtloop_cf.cpp:103)
24  com.prusa3d.slic3r/           	0x00000001016eb6c4 wxCFEventLoop::OSXCommonModeObserverCallBack(__CFRunLoopObserver*, int, void*) + 72 (evtloop_cf.cpp:75)
25  com.apple.CoreFoundation      	0x0000000197871cc8 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
26  com.apple.CoreFoundation      	0x0000000197871b14 __CFRunLoopDoObservers + 572
27  com.apple.CoreFoundation      	0x000000019787117c __CFRunLoopRun + 1052
28  com.apple.CoreFoundation      	0x00000001978705e8 CFRunLoopRunSpecific + 600
29  com.apple.HIToolbox           	0x000000019f78b2a0 RunCurrentEventLoopInMode + 292
30  com.apple.HIToolbox           	0x000000019f78b014 ReceiveNextEventCommon + 552
31  com.apple.HIToolbox           	0x000000019f78add4 _BlockUntilNextEventMatchingListInModeWithFilter + 72
32  com.apple.AppKit              	0x000000019a05f480 _DPSNextEvent + 836
33  com.apple.AppKit              	0x000000019a05de20 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1292
34  com.apple.AppKit              	0x000000019a04fcac -[NSApplication run] + 596
35  com.prusa3d.slic3r/           	0x00000001015a8d0c wxGUIEventLoop::OSXDoRun() + 172 (evtloop.mm:304)
36  com.prusa3d.slic3r/           	0x00000001016ec008 wxCFEventLoop::DoRun() + 44 (evtloop_cf.cpp:331)
37  com.prusa3d.slic3r/           	0x00000001016307b0 wxEventLoopBase::Run() + 152 (evtloopcmn.cpp:90)
38  com.prusa3d.slic3r/           	0x00000001015fcd44 wxAppConsoleBase::MainLoop() + 180 (appbase.cpp:380)
39  com.prusa3d.slic3r/           	0x00000001015fc978 wxAppConsoleBase::OnRun() + 36 (appbase.cpp:301)
40  com.prusa3d.slic3r/           	0x00000001013528a8 wxAppBase::OnRun() + 60 (appcmn.cpp:335)
41  com.prusa3d.slic3r/           	0x0000000101553488 wxApp::OnRun() + 48 (app.cpp:364)
42  com.prusa3d.slic3r/           	0x0000000101664a38 wxEntry(int&, wchar_t**) + 280 (init.cpp:507)
43  com.prusa3d.slic3r/           	0x0000000101664c20 wxEntry(int&, char**) + 72 (init.cpp:519)
44  com.prusa3d.slic3r/           	0x000000010097bdf8 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 292
45  com.prusa3d.slic3r/           	0x00000001002b3ea0 Slic3r::CLI::run(int, char**) + 16320
46  com.prusa3d.slic3r/           	0x00000001002b8728 main + 204
47  libdyld.dylib                 	0x0000000197791450 start + 4

Unfortunately I don't have time to dig in figure out a fix...
If more data is needed or I have a bad build because something is not up to date. Please let me know.

jmd73, 2021-08-12:
Probably same as mine (#6794). It is probably fixed with 7f2ad14.

bubnikv, 2021-08-12:
@justoeu Would you please retest with the current master? Thanks.

justoeu, 2021-08-12:
@bubnikv Sure... Im cloning now the master version and after compile I confirm if its solved.

justoeu, 2021-08-12:
Hi @bubnikv I retest using this new version **version_2.4.0-alpha0-1044-g2be10569e**

Using the same model inside in this issue, and make 2 scenarios: Using 0.2mm profile and 0.15mm profile.

 - Using the 0.2mm profile its perfect;
 - Using the 0.15mm profile crash occurs.

jmd73, 2021-08-12:
I can confirm this crash. It is not the same as my previous issue that was solved.

If I import OP's attached object, switch to Preview.
Then activate "Thick bridges", crash when re-slicing.
It crashes at line 312 in SinkingContours::Update()
            const TriangleMesh& mesh = GUI::wxGetApp().plater()->model().objects[m_parent.object_idx()]->volumes[m_parent.volume_idx()]->mesh();

If I use "Place on face"-gizmo and select a face on the bottom, It does not seem to crash. Only when loaded and sliced in it's default position.

I hope the extra info helps. I tried to save a .3mf, but I couldn't open it again, so I didn't attach it here.

jmd73, 2021-08-13:
@bubnikv Thanks for fixing the loading of 3mf.
@lukasmatena Maybe this info is for you, since you fixed the previous crash?
More info for this crash.

Load attached 3mf (rename from zip to 3mf).
After loading and slicing if background slicing if active, press CTRL+6.
Then press CTRL+2
Activate Thick bridges.
Crash as stated above.

[Vestido_Filha_01_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/6983927/Vestido_Filha_01_fixed.zip)

This is the shortest way I can make to trigger this crash.

jmd73, 2021-08-29:
@bubnikv , @enricoturri1966 I can confirm that at least my attached project above does not crash any more using the same steps as before.

@justoeu Can you also confirm so this issue can be closed?