prusa3d/PrusaSlicer #6798: Printer profile gone again and print settings gone again

sw3de, 2021-08-13:
Sorry for my poor language skills;

### Version
2.3.3+x64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
macOS Big Sur Version 11.5

### 3D printer brand / version + firmware version (if known)
MK3S with latest FW (can't check at the moment)

### Behavior
When I first installed PS 2.3.3+ when it was released most of my print profiles disappeard so I had to re-do them. I thought to myself that it was because it added profiles from addnorth that I had alot of since that is my go to brand. Then it worked well until just now.
I had added a new printer-profile, new filament and a new print-setting.
I have used them for almost two days, exiting the program and opened it lots of times.
Now I changed the name on my printer-setting and saved a new print-profile with a new name and deleted the old.
I exited the program and when I opened it again all three settings to the left showed something like 'default-fff' can't remember exactly.

I had a printer-setting called: Persefone - Prusa i3 MK3S - RIGID X - 70
I changed it to: Persefone - Prusa i3 MK3S - Rigid X - 70

I had a print-setting called: addnorth - RIGID X - 0.20mm
I changed it to : addnorth - Rigid X - 0.20mm

The filament-setting is gone all together, it was called something like: addnorth - RIGID X

My ini files:
[Arkiv.zip](https://github.com/prusa3d/PrusaSlicer/files/6983233/Arkiv.zip)

rtyr, 2021-08-17:
Could you please zip whole data directory and attach it here? Profiles in the bundle you provided seems ok (and are visible in the PrusaSlicer), but I would like to see your data directory.

[image]

sw3de, 2021-08-17:
[Arkiv.zip](https://github.com/prusa3d/PrusaSlicer/files/6998077/Arkiv.zip)
Here it is :)
Sorry if I wrote in anger, but I got very frustrated.
Apart from these two instances (that I wrote before) it has worked flawlessly.
I now have started taking snapshots, which of course always is a good practice.

Thanks for looking into this and I hope this can help others as well.

rtyr, 2021-08-17:
Thanks. This archive is in the state, where some of the profiles you mentioned are missing?

sw3de, 2021-08-17:
I don't understand what 'This archive is in the state' means?
The profiles that got lost I had to re do.

rtyr, 2021-08-17:
OK, just wanted know, if you still don't see those profiles.

I think the issue could be caused by the fact, that you saved a copy of the profile with the same name, but with different cases (RIGID/Rigid). We will investigate what could happen, what operating systems are affected and how to prevent it.

sw3de, 2021-08-17:
Thanks!
Let me know if I can be of any more help.

sw3de, 2021-08-17:
It happened again. I had not made any changes after I opened PS per your request and opened the configuration folder, that I zipped up and posted here.
Just now when I opened PS the print-settings for my Colorfabb nGen settings are gone. I made a snapshot after making that print-setting, but it is not there either.

In this order it happened: (I did use PS to slice some prints in between)
a) I made a filament-setting for Colorfabb nGen, and took a snapshot
b) I made a print-setting for that same filament and took a snapshot
c) Now that latest snapshot does not have the print-settings for colorfabb and starts with '- default -' as print-setting

As you can see on my screenshot the last snapshot states that I'm using my new Colorfabb print and filament settings.. (but the print-setting for Colorfabb isn't there anymore when I load that snapshot)
[image]

sw3de, 2021-08-17:
Just in case I made any changes since I zipped my configuration folder a few hours ago, I made a new one.
[Arkiv.zip](https://github.com/prusa3d/PrusaSlicer/files/6999413/Arkiv.zip)

bubnikv, 2021-08-24:
> And one more thing, if it can have anything to do with this; I am using Superslicer at the same time that is configured for another printer, I don't even have my Prusa set up with Superslicer. Is it possible that that can interfere somehow, that they share some data that gets corrupted?

That is certainly possible. It is equivalent to pour diesel into your gasoline car or vice versa.

bubnikv, 2021-08-24:
Both Windows and OSX file systems consider file names differing only in case as equal. We have to admit we did not cover that behavior well in PrusaSlicer. We need to update PrusaSlicer to disallow what you are doing - we shall disallow saving a profile under a name, which is equal to some existing profile name using a case insensitive compare.

Now why you are losing your profiles when you are doing this kind of operation we don't know. The profile overrides the existing one, that is for sure.

bubnikv, 2021-08-24:
It looks as if we will need to consult with the file system for file name collisions. With localized file names it is quite difficult to ensure that we perform a case insensitive name compare the same way the file system does. For some languages a case insensitive string compare is a very complex thing. I am leaving the job to @YuSanka

closed 2026-09-29 by prusaslicer-bot[bot]