prusa3d/PrusaSlicer #6801: Draft shield shouldn't cut the brim

wavexx, 2021-08-14:
### Version
version_2.4.0-alpha0-1079-g0e0c52ade (master as of today)

### Behavior
In the newest iteration of the brim/skirt/draft-shield options, enabling the "draft shield" along with a brim wide enough, results in the brim being cut by the draft shield.

[image]

This is exactly the same behavior we had previously in #3692.

The primary purpose of the brim is to provide a continuous surface for adhesion, and as a result should it always have priority. The draft shield should be built on top of the brim. On the first layer, the logical operation is currently result=brim-draft, but should be draft-brim.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[BrimCut.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6986527/BrimCut.3mf.zip)

rtyr, 2021-09-20:
From 2.4.0-alpha1 changelog.

> When draft shield is used, skirt loops are printed uninterrupted and brim loops are cut. While it may make more sense to do it the other way round, shortening skirt is unacceptable because of "min_skirt_length" parameter and possible usage for nozzle purging.

Skirt is of often used for priming/purging on multi-material printers.

wavexx, 2021-09-20:
Yes, I fully agree. However when the brim is used, I would prefer to enforce that the brim has extra perimeters added to match min_skirt_lenght. The brim can definitely perform the priming function in addition to adhesion (I generally disable the skirt when printing a single brim object).

closed 2026-09-29 by prusaslicer-bot[bot]