prusa3d/PrusaSlicer #6804: Thin lines ignored/cut

rkarlsba, 2021-08-14:
### Version
2.3.3

### Operating system type + version
macOS 10.15.7

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
Thin lines get cut. See https://karlsbakk.net/tmp/reklame-forbudt1a.stl https://karlsbakk.net/tmp/reklame-forbudt1a.png

The problem varies with layer height. Slicing for 0,4mm nozzle here. This works fine with Cura.

#### Project File (.3MF) where problem occurs
https://karlsbakk.net/tmp/reklame-forbudt.3mf
[reklame-forbudt.zip](https://github.com/prusa3d/PrusaSlicer/files/6986672/reklame-forbudt.zip)

bubnikv, 2021-08-20:
You may enable "detect thin walls"

rkarlsba, 2021-08-20:
already enabled

closed 2026-09-29 by prusaslicer-bot[bot]