prusa3d/PrusaSlicer #6805: Feature Request: improve View Rotation when zoomed in

Jaxx005, 2021-08-14:
### Version
Version 2.3.0+win64

### Operating system type + version
Windows 10 Pro N
Version 10.0.19042 Build 19042

### 3D printer brand / version + firmware version (if known)
Standard factory Prusa MK3S (purchased 2020).
Firmware version: 3.9.3-3556

### Behavior
The centre of View Rotation does not alter when zoomed in to a feature, which causes the object to 'rotate' out of view.  This makes it very diffcult to view small features from different view directions. 

##Example:
Zoom in to a feature (so the object is not
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

###Feature Request
Allow view rotation (and possibly zoom?) about a selected point of interest - espectially when zoomed in.   Perhaps the default could be the centre of the (zoomed) view, on the build plate? This would me similar to CAD programs.

stephen304, 2026-09-02:
I feel like this has gotten somewhat worse but also slightly better in different ways since the 3.0 preview. I noticed before that while right click pan projects the rotate origin to the build plate, middle click pan would only translate the rotate/orbit origin, which made a nice workaround to control where the rotate origin is simply by using middle click instead of right click.

While right click pan now seems to not project the orbit origin anymore, neither right click nor middle click pan seems to apply a translation to the rotate origin, so that workaround behavior is now lost. It seems the rotate origin doesn't seem to stay in the middle of the screen anymore after panning. On the bright side, the rotate origin being projected to the bed doesn't happen anymore, but the loss of the middle click workaround is less than ideal.

stephen304, 2026-09-02:
As an example, here's what panning with middle click looks like in 2.9 - when panning with middle click, the rotate origin is simply translated in the direction of panning, without changing the depth. This makes it possible to manipulate the rotate origin somewhat intuitively, you can see I'm able to travel along the edges of the model and rotate around the corners:

https://github.com/user-attachments/assets/e770c2fd-c47a-459b-99c3-33bffa6ceb76

Here's what middle click pan looks like in 3.0.0-alpha11, even when I pan along 1 axis, the rotate origin ends up being somewhere completely different. It feels impossible to get the rotate origin to be on the corner of the model for example:

https://github.com/user-attachments/assets/84d5ed86-c095-41d8-9306-c3896e0db50c