prusa3d/PrusaSlicer #6806: Import 3MF from Fusion 360

KnatteAnka, 2021-08-14:
### Version
Version 2.3.3+Win64

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
All printers

### Behavior
Fusion 360 is now supporting .3MF files and its the default export option.  
But when importing a .3MF file it gets the name of the first body.   
And not the name of the File imported at least for files exported from fusion 360.  

* What this does it that when i export i get a non descriptive name of the GCode file and its hard to know what it is.

### Expected behavior
* A: Importing 3MF file with set parameter for name it takes that name
* B: Importing 3MF file without  set parameter for a name it uses the file name of the complete 3MF ex "Name Sign Walter v1"

#### Project File (.3MF) where problem occurs
#### Example current behavior:
[image]

#### Base example file used.
[Name Sign Walter v1.zip](https://github.com/prusa3d/PrusaSlicer/files/6986748/Name.Sign.Walter.v1.zip)

neophyl, 2021-08-14:
Sure this is a fusion 360 issues ? As that determines what it names the contents of its 3mf.

KnatteAnka, 2021-08-15:
part ya but what i can find it do not declare any name for the package so then it should be the name of the file.
Same behavior as all other STL files....

neophyl, 2021-08-15:
I think its because you have not actually opened a 3mf file in PS.  When you open a new project and then import objects the default name of the project when you use the save option defaults to the first object  you imported.  As the project hasnt been saved yet and doesn't have a name PS has to do something like that or leave the name blank for you to fill in.  Its being helpful.

When you import from F360 you aren't opening a 3mf file, you are IMPORTING the contents into a blank PS project.  As your objects are called Body (as that is what F360 has named them) then it follows exactly the same process.  Its the difference between opening a 3mf which already has a name and importing the contents of a 3mf or other archive source.  So your requested change would be likely to break the existing program flow.  

What happens when you OPEN a f360 3mf instead of importing it ?

KnatteAnka, 2021-08-16:
the same happens when opening or just importing + sign 
if i compare with a STL it get the base name from the filename and adds a number to each object if separated.

so at least for it being a valid option to use 3mf file in prusaslicer it needs for me to have if imported as one object set the obejct name to file name and not to the first body in the file.
current setup i can accept if opening/importing a file and have as multiple seperate parts.

But best would be if it can be set so i can set it to %Filename%_%BodyName%

rtyr, 2021-09-02:
Fixed in 2.4.0-alpha1. Closing.

closed 2021-09-02 by rtyr