prusa3d/PrusaSlicer #6808: Extruder Gcode output off-by-one issue

mariusstrom, 2021-08-15:
### Version
2.3.3-win64

### Operating system type + version
Win 11 build 22000.120

### 3D printer brand / version + firmware version (if known)
Prusa Mk3S + MMU2S

### Behavior
Using default i3 MK3S and MMU2S profile, the extruder definition is off-by-one. When selecting extruder 3 on a part, the GCODE specifies `T3` when it should specify `T2` (since the printer's extruder count is 0-based (e.g., T0 would be extruder 1). The default Start G-code specifies
```
; select extruder
T[initial_tool]
```

rtyr, 2021-08-15:
Please provide your 3MF project file, where the problem occurs.

Just tried it with simple model. Extruder 3 produced T2 command correctly.

mariusstrom, 2021-08-16:
@rtyr hmm, went to repro it on a brand new STL and it's ... behaving right. Grr.

Unfortunately I don't use 3MF at all today. I guess this is no-repro after all.

closed 2021-08-16 by mariusstrom