prusa3d/PrusaSlicer #6809: Slicer bug, Bridge fill should be solid infill  at 0.35 layer height

clearbucketLabs, 2021-08-15:
[](url)### Version
_Version of PrusaSlicer used goes here_
2.3, 2.3.3

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
When using a 0.35 mm layer height, the slicer puts solid infill (at least on this model) as bridge infill.   When switching to 0.3 layer, slicing and then switching back to 0.35 and re-slicing, it's correct.  closing and re-opening the slicer re-produces the issue again.     the sliced behaviour also matches the printing behaviour which is how i knew. as it was going slower than expected. 

_Is this a new feature request?_
No
#### Project File (.3MF) where problem [occurs]
[mccube v62.zip](https://github.com/prusa3d/PrusaSlicer/files/6988793/mccube.v62.zip)

[image]

n8bot, 2021-08-15:
I've seen this too on 2.4.0 alpha0. Lots of bridging infill areas that don't seem to make sense.

rtyr, 2021-08-16:
Your settings seems a little bit odd. 0.35mm layer height, 0.45mm extrusion width and 0.4mm nozzle. Judging from 6810, you are actually using 0.6mm nozzle.

clearbucketLabs, 2021-08-16:
@rtyr those are the default slicer settings.  I swapped to a 0.4.  easy to change.

closed 2026-09-29 by prusaslicer-bot[bot]