prusa3d/PrusaSlicer #6811: Shortened Filename

thesec, 2021-08-16:
### Behavior
Exporting gcode results in a file name by the STL's filename plus a suffix including printer, filament and print settings ... each separated by an underscore ... by default.

### Wanted Behavior
Allow to export with the STL's filename only by default.

_Is this a new feature request?_
Yes.

rtyr, 2021-08-16:
It is default setting in print profiles for Prusa printers. You can change it in Output options.

[image]

rtyr, 2021-08-19:
No response, possible duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3214. Closing.

closed 2021-08-19 by rtyr