prusa3d/PrusaSlicer #6814: Support Generator "Detect bridging perimeters" message appears on app startup

MShayOffice, 2021-08-17:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 Home

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK2S

### Behavior
On the "Plater" tab, I've selected "Supports: Everywhere" and selected No on the "Detect bridging perimeters" prompt (as it has never worked as well in my prints as the extra supports have).

Then, on the Print Settings tab, I went and saved this as a new User Preset, to avoid having to enable supports and disable bridging manually every time I slice.

After saving that, every time I open PrusaSlicer, the "Detect bridging perimeters" confirmation box appears before the PrusaSlicer application window even opens.

[image]

This is the .ini file for the User Preset causing the issue. It's been renamed to a .txt so that Github would accept it.
[0.10mm DETAIL - Supports.txt](https://github.com/prusa3d/PrusaSlicer/files/6996711/0.10mm.DETAIL.-.Supports.txt)

The expected behaviour is that the application will load normally, without this confirmation box, and won't raise this prompt every time a user preset has "detect bridging perimeters" already turned off.

n8bot, 2021-11-26:
This appears to be broken again in master branch. I keep getting notifications every startup about the support settings. 2.4.0 beta1 downloaded from github does not present the problem with identical steps taken to reproduce.

n8bot, 2021-11-26:
I'm such a newbie, I can't understand for the life of me how this permanently remembers that it queried a profile. Like, it worked perfectly at release of 2.4.0 beta1. How? Is it saving the queried state somewhere? To the profile ini? I couldn't see it.

n8bot, 2021-11-27:
Confirmed now for sure fixed for now.

n8bot, 2021-12-06:
😭 https://github.com/prusa3d/PrusaSlicer/issues/7301#issuecomment-987265446