prusa3d/PrusaSlicer #6815: Update Model/s on File Change

thesec, 2021-08-17:
### Behavior
If a file change to a model is detected ... ask the user to update the build plate with the changed file ... and update the model if confirmed.

_Is this a new feature request?_
Yes, actually not new but already in w/ other slicers for some time ;)

fortinmike, 2022-02-08:
I came here to add an issue requesting exactly this. It happened a few times that I re-exported a mesh from Fusion 360 and proceeded to forget to reload it in PrusaSlicer, printing the previous, slightly wrong model.

Having PrusaSlicer pop a small banner (similar to the "slicing" and "eject" ones) with a suggestion to reload the model when it detects a change on disk would be great!

bubnikv, 2022-02-09:
File watching APIs are platform dependent, see

How do I make my program watch for file modification in C++? - Stack
Overflow
<https://stackoverflow.com/questions/931093/how-do-i-make-my-program-watch-for-file-modification-in-c>

It is doable by polling, but that consumes computer resources
unnecessarily. To do it right, we would have to use the platform dependent
APIs, possibly using
fswatch (emcrisostomo.github.io) <http://emcrisostomo.github.io/fswatch/>

bubnikv, 2022-02-09:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6363

closed 2022-02-09 by bubnikv