prusa3d/PrusaSlicer #6816: Prusa Slicer crash 2.3.3

pbarsanti1, 2021-08-18:
### Version
2.3.3+win64
build 202107161027
_Use `About->About PrusaSlicer` for release versions_

### Operating system type + version
_What OS are you using, and state any version #s_ windows 10, 64 bit, build 19043
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
PrusaSlicer
Version:   2.3.3+win64
Build:     PrusaSlicer-2.3.3+win64-202107161027

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 17,037MB
OpenGL installation
GL version:   4.6.0 NVIDIA 466.11
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 960M/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 306 lines cut ...]
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S+, Stock printer, Firmware 3.10.0-4481

### Behavior
I'm trying to print multiple pieces of the same file. 10 to 20 clips and when I hit slice now it starts to process then crashes my computer. 
[coin_jam_clip_54mm.zip](https://github.com/prusa3d/PrusaSlicer/files/7007513/coin_jam_clip_54mm.zip)

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

fe60, 2021-08-19:
Slicing works on my machine. Main difference between yours and mine: graphics.
Does the operating system crash (blue screen, hangs, ...) or PrusaSlicer?

Just check https://github.com/prusa3d/PrusaSlicer/issues/2939#issuecomment-532563033 to analyze if it's an issue with the graphics driver.

pbarsanti1, 2021-08-20:
HI 

It does seam like it locks up my graphics card. I have a geforce gtx 960m in my Asus laptop. I can slice and export 1 or 2 pieces before it crashes. I have to hard reset my computer when this happens. Ill try changing some of the settings for my graphics card.

Just check #2939 (comment) [https://github.com/prusa3d/PrusaSlicer/issues/2939#issuecomment-532563033] to analyze if it's an issue with the graphics driver.

bubnikv, 2021-09-25:
Do you have the same issue with PrusaSlicer 2.4.0-alpha2?

pbarsanti1, 2021-09-30:
I'll install it today and try it out. Thanks for letting me know an alpha build is available to test.