prusa3d/PrusaSlicer #6818: "Remembers" that SD card is read-only, even after ejected, unlocked, re-inserted.

bkgoodman, 2021-08-18:
### Version
2.3.3+Win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
I insert an SD card that is LOCKED. When I go to save something to it PrusaSlicer says something like "You cannot write to that device". When I eject the device, unlock it and re-insert, PrusaSlicer still will not allow me to save to it (same error message), even though it is truely no-longer read-only, and Windows will let me copy to it. (Yes, PrusaSlicer did "see" that it was ejected and re-inserted). Reproduces consistently.
No

kocikdav, 2021-08-19:
Hello, would you provide exact error message that PrusaSlicer gave you during exporting Gcode?

jmd73, 2021-08-20:
Hello @kocikdav.
I just tried it to help debug the issue, same windows version and PrusaSlicer version.
Detailed steps:
Insert a locked SD-card in computer.
Slice a file.
Press the export to SD-card button (image below this text).
[image]

Press Save in the file requester, and you get the error below (my SD-card is E:)
[image]

Press OK.

Remove SD-card (the export to SD-card button disappears in PrusaSlicer) and write-enable it.

Insert SD-card again. The export to SD-card button re-appears.

Press the button and the same message as before appears.
(Do not remove the SD-card again)
Close PrusaSlicer and restart it again. Open a file and slice it. Press the export button and choose save in the file-requester.
The file saves and you don't get the message that you can't write to the SD-card.

So it seems that PrusaSlicer retains the "write protected" state of the SD-card, even if the card is removed.

I will check the code if I get the time tonight when I get home from work.

I hope my message clears things up, and does not cause more confusion.

kocikdav, 2021-08-20:
Unfortunately this seems to be a Win32 problem. If you try with for example Notepad, you will see it does exactly the same. There is not much we can do about this.  Maybe report to Microsoft..

closed 2021-08-20 by kocikdav