prusa3d/PrusaSlicer #6819: full fan speed layer ma

goddamnmaddog, 2021-08-19:
Version 2.3.3+win64 on Win 10 Pro 64bit

### Behavior
I try to print an object in vase-mode. The upper side of the object is not printing nice. The vase is long and big, g-code has 103601 "layers"
In Filament-Settings-->Cooling--> I try to activate cooling above layer 92000, but input field can only accept values up to 1000.

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-08-19:
That is not how this setting works. When set higher than 'Disable fan for the first [n] layers', the fan speed will be gradually ramped up from zero to maximum in that layer range. Ramping up the fan speed slowly can prevent rapid cooling of print bed and hotend followed by thermal runaway and other related issues. So this setting is mainly about gradual ramp up of the print fan at the beginning of the print job.

**Example**
Fan is set to 80%. Fan is disabled for 1st layer, full fan speed at layer is set to 3.

[image]

What it does:
1st layer: 0% fan speed
2nd layer: 40% fan speed
3rd layer: 80% fan speed. (full speed)

As you can see on this example, usual values are nowhere near to the limit (1000).

If you need to activate print fan at certain level of the vase, you can do that by adding custom g-code command using vertical slider (right click on "+" icon). It is `M106 S255` for 100% fan speed (range is 0-255).

[image]
[image]

goddamnmaddog, 2021-08-19:
Thank you for that very helpful explanation.

rtyr, 2021-08-19:
Can we close this issue?

closed 2021-08-20 by goddamnmaddog