prusa3d/PrusaSlicer #6821: [BUG] Selecting `spiral vase mode` incorrectly disables `Top solid infill` extrusion width value entry field

fiveangle, 2021-08-19:
### Problem
When `Spiral vase mode` is selected, `Print Settings : Advanced : Top solid infill` extrusion width value entry field is disabled, making adjustment of the top layer of the center "solid" base area of the model impossible, even though the value entered into the unchangeable grayed-out field is used when slicing.

The extrusion width for the top layer of the center "solid" portion of the spiral vase mode "base" is extruded at the `Top solid infill` value set in `Print Settings : Advanced : Top solid infill` prior to selecting `Spiral vase mode`.

The continuous spiral layer above this "base" is extruded at the `External perimeters` extrusion value so is not tied to `Top solid infill` extrusion width and so adjusting `Top solid infill` should be allowed when `Spiral vase mode` is `true` in order to tune the top layer of the center "base" portion of the spiral vase model.

### Behavior
- Configure model for Spiral vase mode
- Observe  `Print Settings : Advanced : Top solid infill`  is disabled and pre-populated with default value
- Slice model
- Observe that top solid layer of center "base" portion of vase has EW of default "grayed-out" value
- Unlesslect Spiral vase mode
- Change `Print Settings : Advanced : Top solid infill`  to a much different size as to be visually obvious in layer preview mode
- Select Spiral vase mode
- Observe  `Print Settings : Advanced : Top solid infill`  is again disabled and pre-populated with newly-set much larger value
- Slice model
- Observe that top solid layer of center "base" portion of vase has EW of the new much larger value "grayed-out" value

### Observed on:
PS 2.3.3
Win 10 latest patches
MK3 3.10

rtyr, 2021-08-19:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4241. Closing.

closed 2021-08-19 by rtyr

fiveangle, 2021-08-20:
That issue is not listed as a bug.  Can you please add the appropriate label ?  I tried to dig into the source but couldn't easily identify where this enforcement is set, or I would have made the PR.  This should take minutes for someone familiar with the architecture to implement so it isn't clear why there is zero interest in resolving.  If the issue is correctly marked as a bug, hopefully it will get the attention it deserves.  It's tedious to have to jump through hoops to accomplish such a basic task.  Thanks again !