prusa3d/PrusaSlicer #6826: [Feature request] Move sidepanel

roaldarbol, 2021-08-23:
### Version
2.3.0+

### Operating system type + version
MacOS Big Sur

### Behavior
I work with a multi-screen setup doing quite a bit of coding, thus I have my external screen rotated 90 degrees to a vertical (long) setting - as do many others I work with. However, using it to work with PrusaSlicer doesn't work very well as the `sidebar` remains fixed to the side. See attached screenshot.

It would be a great feature to be able to move the sidebar to the bottom instead (as most IDE's can do, think e.g. VS Code) - it would allow a lot more workspace. 
Anyways, thanks for all the great work you've put in here - overall I'm loving working with PrusaSlicer!

[image]

kocikdav, 2021-08-23:
Have you tried Preferences->GUI->Show sidebar collapse/expand button? Its not exactly moving the sidebar to another side but it will give you more space.

roaldarbol, 2021-08-23:
I have, but I'd still like to have access to the sidebar. As I mentioned, it's a feature request so the aim would be to make it a better user experience on vertical screens without loosing the at-hand functionality. :-)

closed 2026-09-29 by prusaslicer-bot[bot]