prusa3d/PrusaSlicer #6827: Better handling of unloaded filaments in PrusaSlicer when using MMU2S

member68, 2021-08-23:
### Version
2.3.3

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S+ with MMU2S

### Behavior
* _Describe the problem_
I hardly ever have my MMU2S loaded with all 5 filaments at once since I don't have a dry box that can hold all 5 filaments. Therefore, most of the time, there are only 2-3 filaments loaded that I use on that day.<br>
Now, when loading any multi-part model, PrusaSlicer doesn't know that I've only loaded extruder 1-3 and associates the parts of the model with all 5 supported extruders. This forces me to change the unloaded extruders one by one to one of the loaded ones. Depending on the model, that takes a few minutes and is quite annoying when you experiment with different models and want to do some rapid prototyping.

* _Steps needed to reproduce the problem_
Add a multi part model to the build plate. PrusaSlicer will assign extruders to the different parts (iterating through them).<br>
* _Expected Results_
PrusaSlicer should not assign the unloaded extruders to model parts.<br>
This could for example be solved by allowing to select an "unloaded" entry in the filament dropdown. PrusaSlicer would not assign this extruder to the model parts afterwards anymore.<br>
Concerning the handling of already assigned unloaded extruders, I see two possible solutions:
1. Already assigned extruders would be given the number of a loaded extruder. Some error handling would be needed though to prevent you from setting all 5 extruders to unloaded.
2. Already assigned extruders wouldn't change automatically. There needs to be a way to bulk change extuders. Something like "change all parts with extruder 3 to extruder 5".

_Is this a new feature request?_
Yes.

#### Project File (.3MF) where problem occurs
[multi_part_benchy.zip](https://github.com/prusa3d/PrusaSlicer/files/7031306/multi_part_benchy.zip)

jmd73, 2021-08-23:
Hello @member68.

You can multiselect objects in the object list (ctrl+click) and then right click on one of the selected objects on the platter and set the extruder there for all the selected objects. That should make it a little bit faster than selecting the extruder for one object at a time.
Example from you 3mf:
[image]

member68, 2021-08-23:
This saves me about 50% of clicking. Thanks a lot 😄 

In that case I think a "select all parts with the same extruder" option would be a great addition.

closed 2026-09-29 by prusaslicer-bot[bot]